cyberbotics/webots #5137: Whether there is a problem with the compatibility of MATLAB 2017b?

sxbug, 2022-08-30:
**Describe the Bug**
Webots cannot start simulation when using Matlab , Matlab was black screen and Stuck .
 If change the E-puck controller to < extern >, and 1. start MATLAB and open launcher.m. Click run ; 2. start webots simulate , it is OK.
If  you had start simulate success by <extern> mode ,  then you can start simulate by <matlab> mode success.

I think whether there is a problem with the compatibility of MATLAB 2017b?

**Steps to Reproduce**
1.Open e-puck_matlab.wbt
2.Click on 'run to simulation'
matlab open and halt(matlab is black)
no any error msg

Expected behavior
Matlab can only be shut down by force.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
No any Msg, Matlab stuck.

**System**
 - webots 2022b / 2023a nightly build
-matlab 2017b 64bit
-win 10 64bit
 - Graphics Card: AMD Radeon RX 9 M370 2GB

omichel, 2022-08-30:
Yes, it could be a problem with MATLAB 2017b. I just tested with MATLAB 2021a and it works nicely.
Can you upgrade to MATLAB 2021a?

sxbug, 2022-08-31:
I have update to 2021a, But could not use too.

 
Creating: C:\Users\apple\AppData\Local\Temp\protofile_matlab_2021a_webots_R2023a.m
Load Library failed.
error use loadlibrary

Unable to preprocess the input file.

The output generated by the preprocessor is: the system cannot find the specified path.

error launcher (line 91 )
      loadlibrary( ...

》
How can I do ? Thanks

sxbug, 2022-09-01:
Matlab is not stuck, and the interface displays the following contents：

Waiting up to 5 seconds for C:\Users\apple\AppData\Local\Temp\webots_matlab_lock to be deleted by another MATLAB instance.
Deleting C:\Users\apple\AppData\Local\Temp\webots_matlab_lock...
Using prototype file: C:\Users\apple\AppData\Local\Temp\protofile_matlab_2021a_webots_R2023a.m

However, the e-puck robot in webots does not anything either

How can I do？

BenjaminDeleze, 2022-09-07:
Oops did not see that it was a duplicate of #5154

closed 2022-09-07 by BenjaminDeleze