cyberbotics/webots #5144: Viewpoint standard views are misleading

stefaniapedrazzi, 2022-08-31:
Currently when using the Viewpoint standard positions this is the result:
* Front/Back view: align with y-axis and visible plane is x-z
* Right/Left view: align with x-axis and visible plane y-z
* Top/Bottom view: align with z-axis and visible plane x-y

These are relative to the world itself and not to a specific object, so X corresponds to "east" and Y to "north".

But given that object uses now the FLU convention, when selecting the "Front" view I would expect the Viewpoint to align with the x-axis and the visible plane to be y-z.
Respectively when selecting the "Right/Left" view I would expect to see the x-z plane.

Is it correct the current implementation or should we change it to follow the FLU convention?
Note that if an object is rotated the views won't be adjusted to the selected object orientation but will still use the world coordinates.

ad-daniel, 2022-08-31:
related to: https://github.com/cyberbotics/webots/issues/4111

closed 2022-08-31 by stefaniapedrazzi

omichel, 2022-08-31:
@stefaniapedrazzi: feel free to go ahead with the implementation of what is proposed in #4111 (that should be pretty easy).