cyberbotics/webots #5154: Whether there is a problem with the compatibility of MATLAB 2021a?

sxbug, 2022-09-01:
Matlab is not stuck, and the interface displays the following contents：

Waiting up to 5 seconds for C:\Users\apple\AppData\Local\Temp\webots_matlab_lock to be deleted by another MATLAB instance.
Deleting C:\Users\apple\AppData\Local\Temp\webots_matlab_lock...
Using prototype file: C:\Users\apple\AppData\Local\Temp\protofile_matlab_2021a_webots_R2023a.m

However, the e-puck robot in webots does not anything either。

Steps to Reproduce
1.Open e-puck_matlab.wbt
2.Click on 'run to simulation'
matlab open 
no any error msg

Expected behavior
Matlab can close by click close button.

Screenshots
No any error Msg

System

webots 2022b / 2023a nightly build
-matlab 2017b / 2021a 64bit
-win 10 64bit
Graphics Card: AMD Radeon RX 9 M370 2GB

omichel, 2022-09-01:
On my system (Windows 10 64 bit, MATLAB R2021a 64 bit, Webots R2022b, NVIDIA GTX 1660 Ti) everything works nicely and I can see the MATLAB user interface with the e-puck camera and processed images.

sxbug, 2022-09-01:
Change E-puck‘s controller to <extern>, then I can start simulate.
But E-puck's controller is Matlab, the webots's console display:
INFO: matlab: Starting controller: matlab -nosplash -nodesktop -minimize -sd "D:/Program Files/Webots/lib/controller/matlab" -r launcher
INFO: e-puck_position_supervisor: Starting controller: "D:\Program Files\Webots\projects\languages\matlab\controllers\e-puck_position_supervisor\e-puck_position_supervisor.exe"
INFO: 'matlab' controller exited successfully.

the Matlab console display:
要开始，请键入 doc。
有关产品信息，请访问 www.mathworks.com。
 
Using prototype file: C:\Users\apple\AppData\Local\Temp\protofile_matlab_2021a_webots_R2023a.m

Thanks

sxbug, 2022-09-01:
Hi
Change E-puck‘s controller to <extern>, then I can start simulate.

But E-puck's controller is Matlab, the webots's console display:
   INFO: matlab: Starting controller: matlab -nosplash -nodesktop -minimize -sd "D:/Program Files/Webots/lib/controller/matlab" -r launcher
   INFO: e-puck_position_supervisor: Starting controller: "D:\Program Files\Webots\projects\languages\matlab\controllers\e-puck_position_supervisor\e-puck_position_supervisor.exe"
   INFO: 'matlab' controller exited successfully.

the Matlab console display:
  要开始，请键入 doc。
  有关产品信息，请访问 www.mathworks.com。
 
  Using prototype file: C:\Users\apple\AppData\Local\Temp\protofile_matlab_2021a_webots_R2023a.m

Thanks

sxbug, 2022-09-01:
changed controller to extern  , It is OK to start simulate.

sxbug, 2022-09-02:
Hi
Is it because matlab does not display callback information? Is it a permission problem? Is there any further test method?

Thanks

sxbug, 2022-09-02:
If I change the controller to extern, everything works nicely and I can see the MATLAB user interface with the e-puck camera and processed images. But now there is another problem. During normal simulation, pause and simulation again will get stuck. When 2017b was used before, there was no such problem.

I can start MATLAB from the icon. It is work fine.

Why?

Thanks