cyberbotics/webots #5163: Simulation not entirely reproducible after a reset

ad-daniel, 2022-09-02:
**Describe the Bug**

1. Open `sojourner.wbt`, keep an eye on the wheels and play
2. Reset, and play again, the middle wheel behaves significantly different from the previous run

The issue is not a regression and present already in R2022a

omichel, 2022-09-02:
This was observed on Linux but it is not reproducible on Windows.

omichel, 2022-09-02:
On Windows, I noticed that the trajectory of the salamander robot in the guided tour is not exactly the same after a reset.

ygoumaz, 2022-09-02:
It is also not reproducible on MacOS.

BenjaminDeleze, 2022-09-07:
Could be related to #4264 even so in that issue I found a bug only if the simulation is using multithreading

ygoumaz, 2022-11-08:
Could also be related to #2658.