cyberbotics/webots #5164: Missing relative textures when created from PROTO wizard

ad-daniel, 2022-09-02:
**Describe the Bug**

There's two separate issues:

1. Create a `CustomBed` from a `Bed` using the PROTO wizard (and select all the fields so the derived one exposes the same parameters as the base PROTO)
   - Since we selected all the fields, a field `field SFNode blanketAppearance PBRAppearance { baseColorMap ImageTexture { url "Bed/textures/duvet.jpg" } }` is created for `CustomBed` too. Now of course this URL is indeed relative to the original Bed, but isn't to the custom locally available one, so the texture will be missing. When creating a PROTO using the Wizard, if the PROTO exposes assets, we should download them to the local folder.

2. Create a `CustomBed` from a `Bed` using the PROTO wizard (and leaving the default fields selected, namely translation, rotation and name)
   - Since now the appearance field is not overwritten by the derived PROTO, the relative URL needs to be resolved with respect to the remote Bed PROTO (base PROTO), which it appears to do, except it uses the disk file rather than the `http` URL: `Unable to find resource at '/home/testr2022b/snap/webots/common/.cache/Cyberbotics/Webots/assets/Bed/textures/duvet.jpg'.`

omichel, 2022-09-02:
> When creating a PROTO using the Wizard, if the PROTO exposes assets, we should download them to the local folder.

Or we could use the web URL instead of the local one.

ad-daniel, 2022-09-02:
A third issue surfaced, if the PROTO from which we derive references additional PROTO as parameters, these need to be declared. To know what these declarations should look like, we open the derived PROTO and copy the declarations from there. This process already takes place however if the cache is cleaned and we're on a distrib, when attempting to open the reference PROTO to search for the declarations webots crashes since this file isn't locally available (isn't cached).

I'll fix the issue in #5169 as well.

closed 2022-09-05 by ad-daniel