cyberbotics/webots #5166: Scenes/Animations/Simulation of large worlds freezes the computer

ad-daniel, 2022-09-02:
**Describe the Bug**

1. Open `village_center.wbt`
2. Create a scene
3. Try to load it. At some point everything seems to freeze and Webots crashes, I assume it's because the computer ran out of RAM (I have 16GB)

Some optimization might be necessary at some point

BenjaminDeleze, 2022-09-07:
I could not reproduce it on my machine and I have also 16GB RAM.

I checked and it took around 6GB of RAM to open the `village_center` scene in Chrome.
However it took around 11(!) GB of RAM to open the same world in Webots.
So the issue may come from elsewhere but I agree that it seems to be an optimization issue.

Do you have the same problem with other worlds? 
Because I opened other large worlds in Webots (desktop version) to see if it was "common" to have a world taking 11GB of RAM and it seems that is the only one.
I tested all large worlds I could think off (`complete_apartment.wbt`. `hall.wbt`, `tiago++.wbt`, `CH_Morges.wbt`, `CH_Vens.wbt`, `city_night.wbt`, `village_realistic.wbt`) and they take around 2-4GB.

ad-daniel, 2022-09-07:
No I only tested that one, got tired after 3 reboots :smiling_face_with_tear: 
Then perhaps the issue was that while loading the scene in the browser I still had the webots instance up, which together went above the RAM limit. Either way, based on those numbers it doesn't seem possible (or rather likely) for me to be able to stream such worlds (as it requires both Webots and the browser to load it)
Optimizing on the Webots side then probably takes precedence

BenjaminDeleze, 2023-06-12:
Duplicate of #3296

closed 2023-06-12 by BenjaminDeleze