cyberbotics/webots #5173: Convert root to base-node broken with relative URL

ad-daniel, 2022-09-06:
**Describe the Bug**

In a distrib, when the conversion takes place, the relative URLs that suddenly get exposed at the world level need to be converted to remote ones.
When converting EVERYTHING to basenode, the process works as intended and they are turned to remote ones.

There's two ways to go about it:
- Simpler but not ideal: do the manipulation manually in the convert function as we do for the PROTO wizard. 
- Better: figure out why certain nodes/urls don't get converted to remote url whereas they do for a full-basenode conversion

stefaniapedrazzi, 2022-09-06:
This issue is referring to images URLS etc or PROTO URLs?
Because I also experienced a couple of different issues with PROTO URLs after converting root to base nodes the `Tractor` PROTO in `boomer.wbt` after relocation:
* "TractorFrontWheel" not found after conversion if "Tractor" PROTO URL is remote
* "ERROR: Missing declaration for 'SickLms291'" after conversion, save and revert if "Tractor" PROTO URL is local and relative

ad-daniel, 2022-09-06:
It's not related but I've noticed some too on the Lincoln, for which I have a patch already, I'll test if that fixes the boomer too.
So, to reproduce it exactly you:
1. open official boomer
2. do change to trigger relocation message and do so
3. convert tractor's root to basenode?

stefaniapedrazzi, 2022-09-06:
Yes.
Then depending if the Tractor PROTO is local (for example edit the name to trigger the local copy of the PROTO)  or the default remote one I get one or the other error.

closed 2022-09-07 by ad-daniel