cyberbotics/webots #5178: Village_center.wbt seems unusually heavy

BenjaminDeleze, 2022-09-07:
**Describe the Bug**
When opening Webots with `village_center.wbt` it takes around 11GB of RAM on my computer.
This is quite huge and is way more than other large worlds that we have in Webots.
`hall.wbt`, `complete_apartment.wbt`, `CH_Morges.wbt`, `CH_Vens.wbt`, `village_realistic.wbt`, `city_night.wbt` all take between 2 and 4 GB.

I guess it is normal that `village_center.wbt` take more memory because I think it is the more complex world we have. However the difference with the other worlds seems exaggerated. 

**Steps to Reproduce**
1. Check the amount of RAM used
2. Open Webots and load `village_center.wbt`
3. See that the amount of RAM used has skyrocketed.

**System**
 - Operating System: ubuntu 22
 - Graphics Card: GeForce RTX 3060
 - 16 GB RAM
 - CPU: AMD Ryzen 9 5900X

BenjaminDeleze, 2022-09-07:
This protos seems particularly ram-intensive:
```
EXTERNPROTO "webots://projects/objects/buildings/protos/Church.proto"
EXTERNPROTO "webots://projects/objects/buildings/protos/ComposedHouse.proto"
EXTERNPROTO "webots://projects/objects/buildings/protos/HouseWithGarage.proto"
EXTERNPROTO "webots://projects/objects/buildings/protos/ModernSuburbanHouse.proto"
EXTERNPROTO "webots://projects/objects/buildings/protos/SmallManor.proto"
EXTERNPROTO "webots://projects/objects/buildings/protos/SuburbanHouse.proto"
```

<details>
<summary>This simple world is heavier than `CH_Morges.wbt`
</summary>

```#VRML_SIM R2022b utf8

EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackground.proto"
EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackgroundLight.proto"
EXTERNPROTO "webots://projects/objects/buildings/protos/Church.proto"
EXTERNPROTO "webots://projects/objects/buildings/protos/ComposedHouse.proto"
EXTERNPROTO "webots://projects/objects/buildings/protos/HouseWithGarage.proto"
EXTERNPROTO "webots://projects/objects/buildings/protos/ModernSuburbanHouse.proto"
EXTERNPROTO "webots://projects/objects/buildings/protos/SmallManor.proto"
EXTERNPROTO "webots://projects/objects/buildings/protos/SuburbanHouse.proto"

WorldInfo {
  title "Village Center"
}
DEF VIEWPOINT Viewpoint {
  fieldOfView 1
  orientation 0.3047758649309186 0.05876108122568597 -0.9506097030268514 2.95016160956142
  position -18.842668237215292 164.3437819444355 175.61315884114194
  near 4
}
TexturedBackground {
}
TexturedBackgroundLight {
  castShadows FALSE
}
Church {
  translation -183.897744093561 86.22361019581533 0
  rotation 0 0 1 6.021385901
}
ComposedHouse {
  translation -234.55994472601103 138.41864143659825 0
  rotation 0 0 1 5.7595865139999995
}
HouseWithGarage {
  translation -252.47414638078823 82.79827097684856 0
}
ModernSuburbanHouse {
  translation -187.31687350924628 160.71335358119143 0
  rotation 0 0 1 6.021385901
}
SmallManor {
  translation -250.03930255699376 124.84566334159042 0
}
SuburbanHouse {
  translation -274.9253700388613 64.14572330958173 0
  name "suburban house(1)"
}
```
</details>

ad-daniel, 2022-09-07:
I've analysed the Chuch proto a bit and despite being rather simple (just 3 shapes), inserting it into an empty world requires 500 MB. I've replaced the IndexedFaceSets with Meshes and even with a simple Box and it still takes nearly as much RAM. So I think it has something to do with the textures, or rather with their loading since in truth they aren't all that big (the biggest is 3MB and is used once)

omichel, 2022-09-09:
Is it a regression from R2022a?

BenjaminDeleze, 2022-09-09:
I just checked and this world was already burning the RAM in R2022a

ygoumaz, 2022-11-08:
Closing this issue as it is a duplicate of #3296.

closed 2022-11-08 by ygoumaz