cyberbotics/webots #5181: Native code library failed to load.

alexvena19, 2022-09-07:
**Describe the Bug**
A clear and concise description of what the bug is.

**Steps to Reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: [e.g., Windows 10, Linux Ubuntu 18.04, macOS Mojave]
 - Graphics Card: [e.g., NVIDIA GeForce RTX 2080 11 GB, AMD Radeon RX 580 8GB, etc.]

**Additional context**
Add any other context about the problem here.

trying to run a Webots program, and the output is 
Native code library failed to load. See the chapter on Dynamic Linking Problems in the SWIG Java documentation for help.
java.lang.UnsatisfiedLinkError: no JavaController in java.library.path: [/Applications/Webots.app/lib/controller/java]

I'm using a MacBook m1 pro, 64-bit version of java.

[image]
[image]

omichel, 2022-09-07:
Which version of Webots are you using?

alexvena19, 2022-09-07:
> 

Webots R2022a.

alexvena19, 2022-09-07:
> Can you try with the latest nightly build of Webots R2022b?
> You can download it from https://github.com/cyberbotics/webots/releases/tag/nightly_5_9_2022
> Webots R2022b introduces native support for M1 macs.
> It could be that the Java version you are using is built for M1 while the libController of Webots R2022a doesn't support M1, thus the error.
> Using Webots R2022b with M1 support should fix the problem.
> Let me know if that works.

Sure I will try that. Which link of these do I download?
[image]

omichel, 2022-09-07:
webots-R2022b.dmg

alexvena19, 2022-09-07:
> webots-R2022b.dmg

Still not working. I only downloaded the webots-R2022b.dmg

[image]

omichel, 2022-09-07:
> Still not working. I only downloaded the webots-R2022b.dmg

Did you remove Webots R2022a and installed R2022b instead?

alexvena19, 2022-09-07:
> > Still not working. I only downloaded the webots-R2022b.dmg
> 
> Did you remove Webots R2022a and installed R2022b instead?

Yes, when I downloaded it it got replaced by R2022b
[image]

omichel, 2022-09-07:
Do controllers other than Java work for you (e.g., C, C++, Python)?

alexvena19, 2022-09-07:
> Do controllers other than Java work for you (e.g., C, C++, Python)?

I just tried the python example, says I don't have it downloaded as this is a new laptop. I only need the java controller to work for using Webots.

alexvena19, 2022-09-07:
> > Do controllers other than Java work for you (e.g., C, C++, Python)?
> 
> I just tried the python example, says I don't have it downloaded as this is a new laptop. I only need the java controller to work for using Webots.

perhaps I am missing something else to download for java? I have the java 17 64-bit version running on this laptop as shown earlier

omichel, 2022-09-07:
Is your Java version compiled for M1, Intel or both?

alexvena19, 2022-09-07:
> Is your Java version compiled for M1, Intel or both?

I'm not sure, I assume M1. I downloaded java 17 from here: https://www.oracle.com/java/technologies/downloads/#java17

omichel, 2022-09-07:
There are several versions there, some are arm64 (M1) and some are x64 (Intel). Which one did you get?

alexvena19, 2022-09-07:
> There are several versions there, some are arm64 (M1) and some are x64 (Intel). Which one did you get?

arm64

alexvena19, 2022-09-07:
> > There are several versions there, some are arm64 (M1) and some are x64 (Intel). Which one did you get?
> 
> arm64
the x64 did not work when downloading on this computer

alexvena19, 2022-09-07:
update: I tried using cpp and it worked perfectly. must be a java issue?

Tom1072, 2022-09-12:
Hi, I have similar problem and it seems like ARM 64 JDK doesn't work well with SWIG. I tried with X86_64 JDK and things work without problem. Here's my jdk version.
```bash
❯ java --version
java 18.0.2.1 2022-08-18
Java(TM) SE Runtime Environment (build 18.0.2.1+1-1)
Java HotSpot(TM) 64-Bit Server VM (build 18.0.2.1+1-1, mixed mode, sharing
❯ /usr/libexec/java_home -V
Matching Java Virtual Machines (3):
    18.0.2.1 (x86_64) "Oracle Corporation" - "Java SE 18.0.2.1" /Library/Java/JavaVirtualMachines/jdk-18.0.2.1.jdk/Contents/Home
```

linde9821, 2022-11-08:
Is there any news regarding this problem?

I currently try to get a setup working with Kotlin and Gradle. And I have the same issue.

```
Native code library failed to load. See the chapter on Dynamic Linking Problems in the SWIG Java documentation for help.
java.lang.UnsatisfiedLinkError: no JavaController in java.library.path: /Applications/Webots.app/Contents/lib/controller/java
WARNING: 'KotlinTest' controller exited with status: 1.
```

My `build.gradle.kts` looks like this:

```kotlin
plugins {
    kotlin("jvm") version "1.7.20"
    application
}

repositories {
    mavenCentral()
}

tasks {
    val fatJar = register<Jar>("fatJar") {
        dependsOn.addAll(
            listOf(
                "compileJava",
                "compileKotlin",
                "processResources"
            )
        ) // We need this for Gradle optimization to work
        duplicatesStrategy = DuplicatesStrategy.EXCLUDE
        manifest { attributes(mapOf("Main-Class" to application.mainClass)) } // Provided we set it up in the application plugin configuration
        val sourcesMain = sourceSets.main.get()
        val contents = configurations.runtimeClasspath.get()
            .map { if (it.isDirectory) it else zipTree(it) } +
                sourcesMain.output
        from(contents)
    }
    build {
        dependsOn(fatJar) // Trigger fat jar creation during build
    }
}

dependencies {
    // Align versions of all Kotlin components
    implementation(platform("org.jetbrains.kotlin:kotlin-bom"))

    // Use the Kotlin JDK 8 standard library.
    implementation("org.jetbrains.kotlin:kotlin-stdlib-jdk8")

    implementation(
        files(
            "/Applications/Webots.app/Contents/lib/controller/java/Controller.jar",
        )
    )
}

application {
    // Define the main class for the application.
    mainClass.set("KotlinTestKt")
}
```

I especially made sure that I use a X86_64 jvm.

daenenk, 2023-03-15:
I'm using a MacBook with Intel CPU. When launching Webots by clicking on the icon in the Applications folder, and trying to use a Java controller, I also got the error `java.lang.UnsatisfiedLinkError: no JavaController in java.library.path: [/Applications/Webots.app/lib/controller/java]`.
The solution for me was to launch Webots from the Terminal, with `open -a "Webots"` (after making sure that in that Terminal `JAVA_HOME` has been set). Then it worked. I tested several JVMs:
- Zulu OpenJDK 64-Bit 1.8
- Eclipse OpenJ9 64-Bit 11.0
- GraalVM OpenJDK 64-Bit 11.0

2Gameboyz, 2023-09-20:
Tried to make Webots work on my M1 macbook.
I tested different java versions and got two different outputs returned:

Error 1:
`Native code library failed to load. See the chapter on Dynamic Linking Problems in the SWIG Java documentation for help.
java.lang.UnsatisfiedLinkError: JavaController (Not found in java.library.path)
`

Error 2:
`Native code library failed to load. See the chapter on Dynamic Linking Problems in the SWIG Java documentation for help.
java.lang.UnsatisfiedLinkError: /Applications/Webots.app/Contents/lib/controller/java/libJavaController.jnilib: dlopen(/Applications/Webots.app/Contents/lib/controller/java/libJavaController.jnilib, 0x0001): Library not loaded: @rpath/Contents/lib/controller/libController.dylib
  Referenced from: <1230A4CD-5754-3EB5-A093-A02362087696> /Applications/Webots.app/Contents/lib/controller/java/libJavaController.jnilib
  Reason: tried: '/Applications/Webots.app/Contents/lib/controller/java/../../../Contents/lib/controller/libController.dylib' (no such file), '/Applications/Webots.app/Contents/lib/controller/java/../../../Contents/lib/controller/libController.dylib' (no such file), '/Library/Java/JavaVirtualMachines/zulu-17.jdk/Contents/Home/lib/server/./Contents/lib/controller/libController.dylib' (no such file), '/Library/Java/JavaVirtualMachines/zulu-17.jdk/Contents/Home/lib/server/../Contents/lib/controller/libController.dylib' (no such file), ‚/Library/Java/JavaVirtualMachines/zulu-17.jdk/Contents/Home/bin/./Contents/lib/controller/libControl
`

I tested the following jvm and jdk:
Error 1 returned:
- Oracle Java SE 64-Bit 18.0.2 x86_64
- Adopt OpenJDK OpenJ9 64-Bit 11.0.11 x86_64 
- Oracle Java SE 64-Bit 17.0.8 arm64
- Oracle Java SE 64-Bit 1.8.0 arm64

Error 2 returned:
- Azul Zulu OpenJDK 64-Bit 17.0.8 x86_64
- Azul Zulu OpenJDK 64-Bit 1.8.0 x86_64
- Adopt OpenJDK 64-Bit 1.8.0 x86_64
- Eclipse Temurin OpenJDK 64-bit 1.8.0 x86_64

koenyskout, 2024-02-14:
I also got Error 2 of @2Gameboyz when trying webots on my Mac M1.
Interestingly, `/Applications/Webots.app/Contents/lib/controller/java/../../../Contents/lib/controller/libController.dylib` mentioned in the error is missing one `../` in order to work. By adding a symlink `ln -s . Contents` in the `/Applications/Webots.app/Contents` folder, everything started working. So probably there is a wrong path configured somewhere in the build.

CoolSpy3, 2026-10-07:
Looks like this was fixed in #6501.

closed 2026-10-07 by CoolSpy3