cyberbotics/webots #5188: Webots crashes undoing template parameter change

stefaniapedrazzi, 2022-09-08:
Steps to reproduce:
1. Add "PicketFence" PROTO to any world
2. Increase "numberOfSegments"
3. Type keyboard shortcut "Ctrl+Z" (undo)

```
Thread 1 "webots-bin" received signal SIGSEGV, Segmentation fault.
0x0000555555ef0eec in WbEditCommand::resetValue (this=0x55557005f210, newValue=...) at user_commands/WbEditCommand.cpp:62
62	  switch (mFieldValue->type()) {
(gdb) bt
#0  0x0000555555ef0eec in WbEditCommand::resetValue(WbVariant const&) (this=0x55557005f210, newValue=...)
    at user_commands/WbEditCommand.cpp:62
#1  0x0000555555ef0e8b in WbEditCommand::undo() (this=0x55557005f210) at user_commands/WbEditCommand.cpp:54
#2  0x00007ffff747b800 in QUndoStack::undo() () at /home/stefi/develop/webots/lib/webots/libQt6Gui.so.6
#3  0x0000555555e64126 in WbUndoStack::undo() (this=0x5555569097b0) at user_commands/WbUndoStack.cpp:62
#4  0x0000555555b99d1f in WbFieldIntSpinBox::keyPressEvent(QKeyEvent*) (this=0x5555567e5790, event=0x7fffffffd210)
    at scene_tree/WbFieldIntSpinBox.cpp:46
#5  0x00007ffff68252ce in QWidget::event(QEvent*) () at /home/stefi/develop/webots/lib/webots/libQt6Widgets.so.6
```

stefaniapedrazzi, 2022-09-08:
In R2022a, the undo functionality is not enabled after changing the template parameter.

ad-daniel, 2022-09-08:
So the desired behavior is to restore how it worked in R2022a or to allow to undo it (ideally without crashing) ?

stefaniapedrazzi, 2022-09-08:
I think there was a reason why the undo was disabled instead of supporting it.
But if we can support it is for sure better.

closed 2022-09-09 by ad-daniel