cyberbotics/webots #5189: Field editor area too big for content

stefaniapedrazzi, 2022-09-08:
On the master branch the field editor is too big and it is not resized to content.
This is really bad because lot of space is added between the elements and very often you also need to scroll to view and used all the fields when there will be enough space to display all the fields without the scrollbar.

[image]

omichel, 2022-09-08:
This problem doesn't show-up on Windows.

omichel, 2022-09-08:
I cannot reproduce it on my Ubuntu 22.04:
[image]

Which world, node, field are you displaying?

stefaniapedrazzi, 2022-09-08:
It still depends on the number of importable externproto.
If you have many of them the size of the field editor is big.

stefaniapedrazzi, 2022-09-08:
Yes, I'm sure I'm using Webots compiled from the master branch.
I have this issue when working the robotti world.
However, it is not always reproducible and I'm now inspecting which actions are triggering it.
Once found I will fix it.

stefaniapedrazzi, 2022-09-08:
Steps to reproduce:
1. open `sumo_interface_example.wbt`
2. Select `Floor.appearance` field, delete the node and add the `Soil` PROTO node
-> a scroll bar appears in the field editor

ygoumaz, 2022-09-08:
I confirm I can reproduce the issue. This affects worlds with a large IMPORTABLE list. The issue can be reproduced by adding any PROTO node to this type of worlds. Best example is `icon_studio.wbt`.

ad-daniel, 2022-09-08:
The previous issue similar to this was related to the fact that we populated the IMPORTABLE panel even when another panel was open. What I suspect is happening here is that by doing some actions (for example in yannick's case adding a PROTO with add-node), a call to `declareExternProto` occurs and in turn the signal `externProtoListChanged` is emitted. This signal causes the contents of the panel to be updated, or in this case, for it to be filled which messes up the panels yet again.

I suspect it might be sufficient to ensure that we trigger this signal only if the `importable` flag is active

stefaniapedrazzi, 2022-09-08:
Yes, the `updateContents` is triggered by other actions when the WbExternProtoEditor is not active.
Thus, the newly added cleanup method is not called just after to fix the editor size.
However,  I think that we no longer need at all to updated the WbExternProtoEditor when the list of declaration changes: see https://github.com/cyberbotics/webots/pull/5193.

PR #5193 (2022-09-09): Fix field editor size update
Fix #5189:
the `WbExternProtoEditor` content was updated when modifying a new PROTO to the externproto list, even if the editor was not visible, and this was increasing the size of all the field editors.
Given the latest changes in the IMPORTABLE vs other EXTERNPROTO declarations, the importable list can currently only be changed by `WbExternProtoEditor` itself and it is not modified by other user actions.
Thus, we can simplify the code and completely remove the notification when the list of EXTERNPROTO changes.
closed 2022-09-09 by ad-daniel