cyberbotics/webots #5214: TexturedBox PROTO parameters changes not working

stefaniapedrazzi, 2022-09-13:
The changes to the `TexturedBox` PROTO `size` field are not applied in the scene until the world is saved and reverted.

stefaniapedrazzi, 2022-11-01:
The issue is that `TexturedBox` PROTO is not flagged as "template" node.
`TexturedBox` itself doesn't contain any javascript code, but the base `TexturedParallelepiped` does.

omichel, 2022-11-01:
I believe there are two ways to fix this:
1. Add the template language line in the `TexturedBox` PROTO and probably we should also add a test that checks if the template language is set properly in all the PROTO files depending on their dependencies.
2. Fix Webots to flag the PROTO as a template language PROTO if any of its dependency has this template language set.

I believe solution 2 is better in the long term as PROTO developers don't have to care about the dependencies of the PROTO they are creating. It is also cleaner that the template language line concerns only the current file and not its dependencies.

stefaniapedrazzi, 2022-11-01:
Yes, I agree.
I'm inspecting how to implement solution 2.

stefaniapedrazzi, 2022-11-01:
This is a regression. It was working properly in R2022a.

PR #5413 (2022-11-01): Fix isTemplate flag not passed from base PROTO
Fix #5214:
this is a regression from R2022a introduced in #5103 where the instruction to pass the `isTemplate` flag from the base to the inherited node has been removed.

Additionally I also updated some debug instructions throwing errors during compilation.
closed 2022-11-01 by stefaniapedrazzi