cyberbotics/webots #5227: Webots animation: labels not automatically reset

stefaniapedrazzi, 2022-09-15:
If in the simulation a label is only added during the simulation run, when the animation is reset to timestep 0 the label in the animation is not deleted.

**Steps to Reproduce**
1. Go to 'propeller.wbt'
2. Add ``wb_robot_step(1000);`` at line 43 in `axial_and_trail_rotors.c` and recompile controller (red label)
3. Record and save the animation
4. When running the animation for the first time, the red label is not displayed at time 0, but after running it for the first time the label never disappears at time 0

BenjaminDeleze, 2022-09-15:
I don't think this is a bug, more one of the actual limitation of the animations:`Node insertion or deletion are also not recorded in the animation file.`

Even so labels are a bit of a special case as they are not node, but that is the spirit: as only modifications are recorded in the animation file, if a label is created during the animation, it will not have a state at time 0 and so webotsJS will not be able to reset (remove) it correctly.

However, for labels it should be quite easy to fix that such that the behaviour is nicer.

BenjaminDeleze, 2022-09-16:
I investigate this issue a bit yesterday and found out that:
- it is not possible (or at least difficult) to fix this problem without inserting a label modification at time 0
  Something like 
```
"labels":"id":2301,"font":"/home/benjamin/webots2/resources/fonts/Arial.ttf","rgba":"255,0,0,0","size":0.07,"x":0.75,"y":0.02,"text":"Altitude: 0.5 m"}]}). 
```
 This will have for effect to turn the label transparent.
 This is needed because when we jump forward or backward in the animation, we read the _previous_ updates and so we need to have an update saying that at time x the label was not visible.

Obviously I would insert such update at time 0 only if the label was not already declared at this time.

I could do it in two different places:
- In webotsJS:
  - Easier to implement.
  - Would also correct previously created animations.
  - Do not increase size of the json to send across the network.
  - Would also work on animation generated by scripts (like the human skeleton one).
  - Add a little overhead in webotsJS.
- In webots:
  - Increase (a bit) the size of the JSON.
  - Add no overhead on webotsJS.

From what I stated above I would prefer to implement it in webotsJS, but do not hesitate if I am missing something.

omichel, 2022-09-16:
I am concerned that we will have to parse the whole animation file to add these initial empty labels, so the overhead will scale with the side of the animation file... And this will affect all the animations, not only the ones with labels which are pretty rare. So, I don't like too much this first option and would prefer the second one, which seems cleaner to me. I agree it would be more difficult to implement (see need to store the animation data in a structure in memory until the animation is complete, add the initial empty labels and save the result).

BenjaminDeleze, 2022-09-16:
I think the overhead will be very small.
I have two solutions to implement it:
- the `json` file already contains a list of `labelsIds` so it would be enough to get the list of ids and to modify the first frame to add the new entry (we do not really need information other than the labelsIds as we want the label to not be visible so we do not care about size, position, etc...)
- We already go through the whole animation file once in order to generate the keyframe. So I could also generate the additional label update in the same loop, which would add close to no overhead.

So if I am not sure the overhead alone should justify our decision

omichel, 2022-09-16:
I see. Thank you for the explanation. So, what you propose sounds good to me.

PR #5236 (2022-09-20): Fix labels reset
At setup we check if all labels are present in the first frame to have a reference point when jumping backward and forward in the animation.

If it is not the case, create a label instruction in the first frame which set the label to transparent (and empty)

Fix #5227
closed 2022-09-20 by BenjaminDeleze