cyberbotics/webots #5228: Problem with animation and viewpoint follow

BenjaminDeleze, 2022-09-15:
**Describe the Bug**
If the viewpoint follow is activated in animation some issues could happen:

- If you move the viewpoint _before_ starting the animation, there will be a wrong jump of the viewpoint when you start the animation.
- When the animation is on loop, when it starts again, the viewpoint is sometime wrong.

BenjaminDeleze, 2022-09-16:
I studied a bit this issue and currently the "correct" behavior is the following:
- When the viewpoint follow is on, the viewpoint position/orientation is reset each time the animation start over.
- When the viewpoint follow is off, the viewpoint position/orientation is never reset automatically.
- When the viewpoint follow is on and we move the viewpoint before starting the animation, webotsJS will try to reset the viewpoint when we start the animation (it happens because currently we reset the viewpoint if follow is on and we are at time 0).

So I think that we should harmonize the behavior such that :
- If we move the viewpoint before starting the animation, it is not reset at start.
- Decide if we want to reset the viewpoint when the animation restart (independently from the viewpoint follow)
  - Yes: because it is the same as a reset.
  - No: the user decided to change the viewpoint so we should not change it and he can use the reset viewpoint button if he wants.

omichel, 2022-09-16:
I would go for the "No" as we don't want to loose the viewpoint set by the user and the user still has the possibility to reset it if she wants to.

PR #5237 (2022-09-20): Fix webotsJS follow viewpoint
Fix #5228 

I tried several things but finally went for a very simple solution:

- If the viewpoint follow is not enabled, we never touch the viewpoint position (except if the user reset it manually).
- If the viewpoint follow is enable, we reset it **but** we preserve the transformations (translation and rotation) done by the user.
  We are force to kind of reset the position otherwise we have some problems with the follow (it does not like big jump).
closed 2022-09-20 by BenjaminDeleze