cyberbotics/webots #5279: Controller gets stuck after running once

correll, 2022-09-28:
**Describe the Bug**
We are working in the "Complete Apartment" demo and observe that Python controllers get stuck after running once. After pausing or restarting, the time will simply not advance. It works again when setting "synchronization" to FALSE, that is the controller seems to get stuck. Multiple students have this problem (not all), and I could not find commonalities. All of them use Windows with a variety of graphic cards (Intel, Nvidia). 

The complete apartment world had some problems with the physics, so I deleted these before shipping to the students. Maybe there are other physics nodes in there that are problematic.

**Steps to Reproduce**

We can share the lab / controller that we ship. Its complete apartment with a Turtlebot Burger in there.

ad-daniel, 2022-09-29:
The controller is direct or external? Also yes, if you can provide a zip file of the exact project that is causing issues it could help debug the problem

ad-daniel, 2022-09-29:
Also, which Webots version are you using?

correll, 2022-09-29:
All the students are using Webots 2022B, all different computers. I asked them to update to Python 3.10. It seems that those who have trouble are having bugs in their python syntax. It shows the error message only once, then they have to restart webots to see additional error messages. This is a problem if they have multiple syntax errors; they fix the first, it does not show additional errors, then need to restart and so on.

omichel, 2022-09-29:
I tried to reproduce this problem on my local machine, but did not succeed. Could you please share a sample python controller that causes the problem?

PR #5284 (2022-09-30): Fix controller restart after crash
Fixes  #5279.
closed 2022-09-30 by omichel