cyberbotics/webots #5281: Prerequisites for web PROTO

ad-daniel, 2022-09-29:
Prerequisites for web PROTO:

1. Field callbacks: all Js nodes need to have "update" methods so that we can link them to parameters. Typically stuff like size, radius, color, translation, rotation of all primitives/nodes should be possible to update with a setter
2. Node completeness: this is more of a discussion, but it's best if it's settled before rather than after. Currently the nodes on the js side are a simplified version of the vrml counterpart (ex: transform instead of Robot). This is fine, but depending of the plans going forward the nodes should be more 1:1.
   - To show joint locations and joint limits: generating this info from Webots is trivial because everything is at hand. Generating this info while decoding the vrml is not trivial (the decoding isn't linear, need to enter into nested PROTO, deal with derived PROTO, keep track of transforms, etc). If the resulting x3d is simplified, nothing persists about the joints. If the structure is maintained, all you need is to find the joint nodes in the node tree
   - If the purpose is just to show the 3d view, a simplified tree is sufficient. If the purpose (later on) is to assemble PROTO, then inserting a transform into a SFNode isn't the same as inserting a Solid. Moreover, many of the fields that have no 3d rendering still need to be there because the user might want to adjust them (camera resolution, motor properties, solid radar properties, ...)
   - PROTO aren't sanitized. There is no distinction between garbage data and expected (supported or not) field values. So even if the info is not used on the js when rendering, we need some form of node prototype to know what to expect and what to skip
   - it doesn't have to be on either extreme (simplify everything or nothing at all). Just having the nodes (Robot, Solid, Joint) on the js side might suffice, even if most of their fields aren't defined. What matters is that we need to decide before what is the philosophy
3. Tessellation: some of the proto define indexfacesets using quads or even polygons (circle arena). Some of this info is hidden behind templating. Either we convert all the proto to only use triangles, or a tessellator is needed

BenjaminDeleze, 2022-09-29:
I think we also need to define what we want to do with web PROTO:

- Replace the documentation: we should display the position of joints, motors, devices and be able to move the joints.
- Modify the parameter of the protos: we should display the header of the proto and be able to regenate it if some value changes (like color, translation, children...)
  - Do we want to support only "simple fields": color, translation, rotation, scale, ...
  - Or do we want to also support inserting arbitrary node in children. 
- Or both.
- Or other things.

Everything should be feasible, but the interface will be way different, and the time needed too

BenjaminDeleze, 2022-09-29:
So we discussed with @omichel and here our conclusions:
The goal of web protos are:

1. To replace the corresponding documentation.
2. To let the user modify the parameter present in the header of the proto

Precisions:

- The sliders for motors and joints are nice and they should be kept. However it does not seems a good idea to have a `.json` generated by Webots because modifying some parameters of the proto could make the `.json` not relevant.
- The user should not be able to modify the body of the proto.
- The user should be able to download the proto with the modified header.
- Some parameters are nodes. For web proto, we will limit the possibilty to other protos and not base node.
  - For example, if a user want to replace the appearance of the door, it can remove it and add another proto appearance (and modify its color) but not add a base PBRAppearance node. Otherwise it will become too complicated.
- I think that, to be the most generic possible, we have to generate a precise `.x3d` file. This will also allow us to generate the sliders of the joints and motors. 
  - What need to be added (in addition to the graphic aspects that we already have) is: joints position and amplitude, motor and sensor position. 
  - However we can drop the parameters of the sensors and motors as they will not be visible on the web.
  - The user should still be able to modify the parameter of sensors (like resolution of the camera) if the camera is in the header of the proto. But we can change it only in the `.proto` as it will have no impact on the `.x3d`.

ad-daniel, 2022-09-29:
> For example, if a user want to replace the appearance of the door, it can remove it and add another proto appearance (and modify its color) but not add a base PBRAppearance node. Otherwise it will become too complicated.

So this functionality (insert SFNodes specifically) should be available for the documentation too or would be implemented only in the context of competitions (with limited library of PROTO components)? A difficulty I can see is with node compatibility, as basically we need to have metadata about all PROTO be readily available (what is their basenode, if it's a slot what is their type, if it's a device it can only be inserted into another PROTO of basetype Robot etc). It's doable, just not sure how fast it will end up being since this needs to be extracted to determine the candidates, on top of actually spawning the selected one.

> I think that, to be the most generic possible, we have to generate a precise `.x3d` file. This will also allow us to generate the sliders of the joints and motors. 

So to be sure you mean having the x3d be complete (i.e. have a robot node, a solid node, etc) but in practice only to "write" (when exporting) the minimal necessary parameters for each of them rather than the full list, correct?

> However we can drop the parameters of the sensors and motors as they will not be visible on the web

Well sorta. It's true the information just needs to be encoded in the exported proto (exported from webots.cloud to a file) but when the user does this change in the interface it still needs to be rendered on the web and the underlying node potentially regenerated. Usually the roads are like that, or the version parameter of a Nao. It's a harmless string as a parameter, but in practice requires rebuilding the nao entirely to render it correctly (the x3d is no longer the same). A user might create a PROTO of a TV, and scale the 3d model of the TV based on the selected resolution.
I think non visual parameters can still have visual side-effects.

But overall I agree with everything

BenjaminDeleze, 2022-09-30:
> So this functionality (insert SFNodes specifically) should be available for the documentation too or would be implemented only in the context of competitions (with limited library of PROTO [...]

You are right, maybe it could be possible to store the base type of each Proto in the DB of webots.cloud. Currently we already parse the proto to find its `description` and `name`. I think that the idea of @omichel was to make it available also for the documentation. Either way I think that this is the last (and maybe the more difficult) step of web proto, so we still have some time to think about the optimal way to implement it, except if it introduces fundamental design choice...

> So to be sure you mean having the x3d be complete (i.e. have a robot node, a solid node, etc) but in practice only to "write" (when exporting) the minimal necessary parameters for each of them rather than the full list, correct?

Yes, I thought having in the x3d: the graphical element (as we have now) and the element needed to build the sliders (so the joints and motors but only with the necessary parameter).
 
> Well sorta. It's true the information just needs to be encoded in the exported proto (exported from webots.cloud to a file) but when the user does this change in the interface, it still needs to be [...]
I think non visual parameters can still have visual side-effects.

You are right. Would it be possible, when parsing the `.proto` file, to build a list of parameters that trigger regeneration?
If this is possible I think it is maybe the best way because it would allow us to:
- regenerate the `.x3d` only when necessary (proto regeneration is triggered)
- otherwise handle the modifications directly in the js node.

ad-daniel, 2022-09-30:
> You are right, maybe it could be possible to store the base type of each Proto in the DB of webots.cloud. Currently we already parse the proto to find its description and name. I think that the idea [...]

Actually it might be sufficient to run the python script we use to generate the proto-list.xml on a nightly basis (CI job) and just get the metadata from that. But yes, still need to setup the rest before, was just thinking out loud

> regenerate the .x3d only when necessary (proto regeneration is triggered)

Yes regeneration was detected and handled in the proof of concept PR, so should work

omichel, 2022-09-30:
We should also keep in mind that this tool should be also able to display any user-contributed PROTOs, so we cannot assume that a script will run in the CI to update some information related with the PROTOs.

BenjaminDeleze, 2022-09-30:
We decided with @omichel, that, at least in a first time, we will only be able to display the PROTOs that can be introduced at the root level (transform, shape, solid, robot, cadshape,...) and not the other ones (geometry, distancesensor,...)

ad-daniel, 2023-02-28:
I think we can close this now, web PROTO is well underway and some of the things mentioned here are either done or no longer relevant

closed 2023-02-28 by ad-daniel