cyberbotics/webots #5282: Controller is crashed in apple silicon

kslee12, 2022-09-29:
**Describe the Bug**

Controller is crashed in my laptop, Chip Apple M1 Pro (macOS Monterey 12.5.1)

The problem can be reproduced as follows(I tried both R2022b(official) and nightly build
).

**Steps to Reproduce**
--------------------------------------------------------------
First, if I run the controller,  it produces error :

1. Run webots.app

2. Open any example with c controller, such as bitcraze with crazyflie.c

3. clean and rebuild crazyflie.c
```
make clean
Clean finished.
make 
# updating pid_controller.d
# updating crazyflie.d
# compiling crazyflie.c (x86_64)
# compiling pid_controller.c (x86_64)
# linking crazyflie (x86_64)
# compiling crazyflie.c (arm64)
# compiling pid_controller.c (arm64)
**# linking crazyflie (arm64)**  <---  verifies linking  arm64 build
# creating fat crazyflie
# copying to crazyflie
Build finished.
```
5. Run simulation, it shows WARNING, and the simulation is stopped  : 
```
INFO: crazyflie: Starting controller: /Users/kslee/webot_ws/projects/robots/bitcraze/crazyflie/controllers/crazyflie/crazyflie
WARNING: crazyflie: The process crashed some time after starting successfully.
WARNING: 'crazyflie' controller crashed.
```
---------------------------------------------------------------

Now, how can we certain that this issue is related with arm 64 build?
I tried to run same procedures with Rosetta(x86_64 emulator)  :

1. Run webots.app with Rosetta2

2. Open any example with cpp controller, such as bitcraze with crazyflie.c

3. clean and rebuild crazyflie.c
```
make clean
Clean finished.
make -j 8
# updating crazyflie.d
# updating pid_controller.d
# compiling crazyflie.c (x86_64)
# compiling pid_controller.c (arm64)
# compiling pid_controller.c (x86_64)
# compiling crazyflie.c (arm64)
# linking crazyflie (arm64)
**# linking crazyflie (x86_64)**  # verifies linking  x86_64 build, emulated by Rosetta 
# creating fat crazyflie
# copying to crazyflie
Build finished.
```
4. Run simulation, it shows expected behavior : 
```
INFO: crazyflie: Starting controller: /Users/kslee/webot_ws/projects/robots/bitcraze/crazyflie/controllers/crazyflie/crazyflie
 Take off! 
====== Controls =======
 The Crazyflie can be controlled from your keyboard!
 All controllable movement is in body coordinates
- Use the up, back, right and left button to move in the horizontal plane
- Use Q and E to rotate around yaw
-----------------------
```

omichel, 2022-10-03:
May be related with #5181.

omichel, 2022-10-03:
I could reproduce the problem. It seems that the version bundled with Webots can execute the arm64 part without any problem. The problem arises only when recompiling a controller program. I found a workaround. Once compiled copy the arm64 version only to overwrite the version generated by the Makefile, from a Terminal:
```bash
$ cd /Users/me/Desktop/crazyflie/controllers/crazyflie
$ cp build/release/arm64/crazyflie .
```
Let me if that works for you.
I believe we need to fix the [Makefile.include](https://github.com/cyberbotics/webots/blob/master/resources/Makefile.include#L547-L550) to address this problem... I will further investigate this.

omichel, 2022-10-03:
Removing [this line](https://github.com/cyberbotics/webots/blob/42aaa7414c9b9dd45dbe0bb11c1173fb2c7eb7ac/resources/Makefile.include#L550) from your `/Applications/Webots.app/Contents/Resources/Makefile.include` file should fix the problem.

Can you confirm this works for you?

kslee12, 2022-10-04:
@omichel, 

I think your solution is correct, since it works for me. 
Now, the simulation shows no error with same procedures that I described before.

I also have another problem related with python controller(installed with homebrew), however this problem should be discussed in another issue. 

Currently, I am just happy to play Webot with cpp controller, Thank you.