cyberbotics/webots #5292: Error creating folder /Users/protos when editing proto file in Mac OS

tuncalie, 2022-10-05:
**Describe the Bug**
Whenever I try to edit a proto file in mac, it first asks me to confirm that it can create a copy of the proto file and then complains with the message "Error creating folder /Users/protos/".

**System**
 - Operating System: macOS Monterey 12.6 (M1 Pro CPU)
 - Graphics Card: M1 PRO

omichel, 2022-10-05:
I just tried on the same configuration as you and it worked fine.
Can you describe precisely what you did? Which world did you open? Which proto did you try to edit? Which folder did you select? etc. Obviously if you select a folder for which you don't have write permission, it will fail...

tuncalie, 2022-10-11:
I was having this issue when my world file was under my user directory without the Webots's project directory structure. I was able to solve this issue by first creating a project directory, and putting my world file under 'worlds' directory.

closed 2022-10-11 by tuncalie