cyberbotics/webots #5298: Lidar ranges (are again) not available after API restart

ShuffleWire, 2022-10-05:
This bug came back in 2022b, it had been fixed before.
https://github.com/cyberbotics/webots/issues/3509

stefaniapedrazzi, 2022-10-06:
I confirm the regression.
The connection to the extern controller code has been completely refactored and during the refactoring it seems that the code to support extern controller reconnection has been accidentally removed.
I will readd it and check that all the previously reported use cases are working again.

ShuffleWire, 2022-10-06:
the issue is actually caused by not calling wb_robot_cleanup() (when my controller crash, I've no way to execute it before it terminate)
If this function is called, everything work, and the next controller connection allows to continue the simulation from the point it left it, and lidar data are received.

stefaniapedrazzi, 2022-10-06:
It turns out that the extern controller is correctly reconnected (apart from an assertion failure raising in debug mode).
The problem is very specific to the lidar code and it is fixed in https://github.com/cyberbotics/webots/pull/5305.

closed 2022-10-10 by stefaniapedrazzi