cyberbotics/webots #5324: After error the controller does not start

laszlo-schaffer, 2022-10-12:
**Describe the Bug**
After an error occurs in the controller (python in this case), it does not start again. In the console it can be seen that the controller is going to be started with the corresponding command. But the simulation is not started, the runtime halts at 0:00:00.

**Steps to Reproduce**
1. Start the simulation, which produces an error in the controller.
2. Reset the simulation
3. Start the simulation again.
4. See that the simulation is halted, the controller wont start either.

**Expected behavior**
After the error is fixed and the simulation is started, it is expected that the simulation will run without the error.

**Screenshots**
In the screenshot it can be seen, that an error occured in the simulation. The error has been fixed, the simulation is started again after a reset, but the controller wont start, the simulation halts.

[image]

**System**
 - Webots 2022b
 - Operating System: Windows 10, 10.0.19044 build 19044
 - CPU:  Intel(R) Core(TM) i7-4770K CPU @ 3.50GHz
 - Graphics Card: Nvidia Geforce GTX 770

**Additional context**
This error did not occur in earlier versions of Webots, like 2022a.

omichel, 2022-10-12:
This bug was fixed in #5284.
Please download the latest [nightly build](https://github.com/cyberbotics/webots/releases) of Webots R2022b-rev1 which includes the patch.
Let me know if that resolves the problem.

laszlo-schaffer, 2022-10-12:
Thank you. Indeed, it solves the problem.

closed 2022-10-12 by laszlo-schaffer