cyberbotics/webots #5326: Convertion from PROTO to Robot don't keep the controller

ShuffleWire, 2022-10-13:
**Describe the Bug**
When converting a Robot PROTO to a regular Robot node, the field "controller" revert back to `<generic>`.

**Steps to Reproduce**
1. Create a new world
2. Create a new robot proto
3. Import it in the world
4. Change controller to whatever (for instance `<external>`)
5. Use "Convert to Base Node"
6. See the message `INFO: The 'controller' field of the robot has been changed to "<generic>".`

**Expected behavior**
controller field should keep is value.

**System**
 - Operating System: Debian 11
 - Webots master@152f170ed0944177e5e0ed98b6bb899be8c97478

ad-daniel, 2022-10-13:
The only difference with https://github.com/cyberbotics/webots/pull/5125 is that if I understood correctly in this case the PROTO is local and presumably the controller too. For remote PROTO we need to keep the current behavior, but if we could identify that the controller is local, then we can avoid turning it into a generic one.

ygoumaz, 2022-10-13:
I even think this only affects controllers set to generic, none or extern, as we already check if the controller is local to the project.

ShuffleWire, 2022-10-13:
TBF, my use case of this issue is first and foremost with extern controller. ie I often have to convert my own robot proto to nodes, and doing so I have to manually reset controller to extern.
additionally, I could miss something, but I don't see any case with an extern (or a none) controller in with being local or node have some importance, the controller being fired up externally, and from the user responsibility, anyway.

closed 2022-10-14 by ShuffleWire