cyberbotics/webots #5330: Issue when connecting extern controller

ShuffleWire, 2022-10-14:
**Describe the Bug**
When connecting a external controller for the first time (after a world reload), the messages 
"Cannot connect to Webots instance, retrying in X seconds" shows up.
It's a regression from 2022a, but it's not so bad, tbf. It's just that it's annoying, and cause delay to start the world (because the delay between retry increase over time, I could spend sometimes a few seconds to way for my world start).
After a reset and restarting the controller, everything work without any messages nor delay.

**Steps to Reproduce**
1. Open a world with a Robot with an synchronized  external controller, and don't start it (ensure it's paused)
2. Run the external controller.
3. See that messages "Cannot connect to Webots instance, retrying in X seconds" are displayed on the controller. That a regression, 2022a made the controller connect to Webots immediately. The controller get stuck on `wb_robot_init()`
4. Press play, then Webots will wait a few seconds for the controller to start, and then run the simulation normally.
5. Reset (not reload) the world + kill the controller 
6. Re run controller, it start immediately, run immediatly `wb_robot_init()` and wait on `wb_robot_step()`. That is the correct behavior.

ygoumaz, 2022-10-14:
This behavior is intended, or at least we already came across this issue and had a discussion with @omichel. I don't remember exactly the reasons why we decided to keep it this way. @omichel may have a complete answer about this.

Anyway, one option that you have is to start Webots in `realtime` mode, so that the controller can directly connect once Webots is ready.

Also, is this a behavior that you encounter only in the last nightly build or is it already present in the R2022b official release?

ShuffleWire, 2022-10-18:
> Also, is this a behavior that you encounter only in the last nightly build or is it already behaving this way in the R2022b official release?

In both version

ygoumaz, 2022-10-21:
Currently the behavior is consistent between intern and extern controllers. Controller executables connect to the simulation only after the play button is pressed. This is also the case for intern controllers launched by Webots. So unless a real use-case for this shows up, we won't change the behavior.

Maybe we can reduce the delay between each connection attempt of extern controllers. Instead of respectively 1, 2, 3, 4, ... seconds we could just try to connect every second to reduce the delay. What do you think?

ShuffleWire, 2022-11-02:
I do think that it's fine to keep the 1,2,3,4...

I only a small inconvenience, if it become a bigger issue, I will open back the issue.

closed 2022-11-02 by ShuffleWire