cyberbotics/webots #5331: Robot windows don't respect device timestep

ShuffleWire, 2022-10-17:
**Describe the Bug**
When reactivating a device from the robot windows, it does that using the timestep of the world, and not the initial timestep of the device. 

**Steps to Reproduce**
1. Open sample world lidar
2. Note that world timestep is 8, and lidar is enabled in the controller using TIME_STEP==32
3. Modify the controller to print in the `while` loop `wb_lidar_get_sampling_period(lidar)`
4. Run the world, see that it print 32 -> OK
5. Open robot windows, override controller, untick lidar activation, the controller then print 0 -> OK
6. Tick back the activation, and controller print 8 -> Not OK, it should go back to 32

**System**
 - Webots : master@270d7f673c6bba7c40c974dbf2826a2188a5260f

stefaniapedrazzi, 2022-10-18:
This is the expected behavior.
The default robot window is just a generic and simple tool to easily visualize the sensors providing basic functionalities but it is not meant to be used to precisely control a simulation.
It is built to work independently from the controller code that is why it was chosen to simply use the world basic time step.

So, this enhancement, even though it is not very complicated to implement, it is not a priority.

ShuffleWire, 2022-10-18:
That fine for me !

closed 2022-10-18 by ShuffleWire