cyberbotics/webots #5332: Device activation keep it previous state even after reset

ShuffleWire, 2022-10-17:
Disclaimer: I'm presenting a behavior that should not be encountered very often, and moreover by using a very special setup... But it still show some inconsistencies that could be addressed.

**Describe the Bug**
NB: The following should extend to any sort of Device, not only Lidar.
I'm using costly Lidar (in term of computing power), and they are not always used. Hence I'm actively toggling them on/off during world execution to save time.
My controller assume that if nothing have been done, the Lidar was deactivated, hence saving comuting ressources. And later, I could activate it when needed.
It work actually that way, but there is a caveat : Here is how : 

**Steps to Reproduce**
1. Open world with a Lidar, and a controller which activate the lidar.
2. Run the controller, lidar work fine
3. Change the controller to not activate nor deactivate the Lidar
4. (in a case of an external controller) Restart the controller, without resetting the simulation : the lidar is still running. It seems ok, as long as we consider that we just "resume" the simulation where we left it. But the controller start with an enabled device...
5. Reset the world, re run the world and the controller, and still the lidar is activated ! That don't seems right, because we gone back to the initial state of the simulation. It could potentially to non reproducible simulation, between reset.
6. Reload the world, re run and see that the lidar is deactivated -> that is correct, but not consistent with the reset behavior

**Expected behavior**
I feel like that I least the reset and the reload should behave the same, and ensure that starting condition are the same each time.

**Current fix**
For the moment I just start by enabling or disabling the Lidar, to ensure the activation status.
I guess it should be the correct way to do it, but I find the reset behavior disturbing, however. Or at least add some lines in the documentation, just to be sure...

To make it more persistent, either a reset should reset activation status to false (IMO the best solution), either call `wb_***_disable()` in each `wb_robot_get_device()` (but could add many code...)

Thanks

ShuffleWire, 2023-09-06:
It's seems to be fixed in 2023b, but I don't know how...
If anyone have a idea before closing, I would be happy :)

omichel, 2023-09-07:
It seems it was fixed in #5889.

CoolSpy3, 2025-05-24:
It looks like this is a thing in R2025a? Although, based on [this comment](https://github.com/cyberbotics/webots/pull/5889#issuecomment-1429632238), it might not actually be an issue. Thoughts on if we should still look into this?

omichel, 2025-05-24:
I tend to agree with @ShuffleWire that this behavior is odd and it looks like a bug. Since devices start-up disabled, the reset functionality should set them in their initial state again, that is disabled.