cyberbotics/webots #5371: GPS initial speed

ShuffleWire, 2022-10-26:
When using a GPS, the first (vector) speed return is not correct.
The explanation is straightforward : 
1) in the code WbPGS.cpp, the initial value of `mPreviousPosition` used to estimate the speed don't seems to be initialized (or maybe the constructor set it to zero by default ?)
2) In the reset function, it's clearly resetted to 0

in both case, that yield to wrong speed estimation when the robot is not actually at the 0,0,0 coordinate at the start, (which is almost always the case, robot are moved anywhere in the world)
The correct way to do it it either :  
 - Initialize this value `mPreviousPosition` to the first position of the robot before starting the simulation, hence, the first speed should be 0,0,0, but it's not that bad. However, I don't know if we have access to this info at that moment.
 - Intialize `mPreviousPosition` to nan, and then do something like
 - 
```
if (!mPreviousPosition.isNan())
    mSpeedVector = (t - mPreviousPosition) * 1000.0 / mSensor->elapsedTime();
  else
    mSpeedVector = WbVector3();
```
I could do the fix, but let me know what is the best way to go. 

In the end, the best way to get the GPS speed would be to get the speed from ode directly, but this could be more work...

omichel, 2022-10-26:
A real GPS usually computes the speed from the change of position only (there is nothing similar to getting the speed from ODE). This means that a GPS needs at least two iterations to compute the speed.
As a consequence of this:
1. It is acceptable to return NAN if the position history is not sufficient. Hence on the first step or after a reset, the GPS should return a NAN value for the speed.
2. The GPS should not get the speed from ODE as this sounds like cheating, e.g., doing something that a real GPS couldn't do. Moreover, if the GPS position is noisy, the speed position should be noisy as well (since its computation rely on the position), which is realistic.

ShuffleWire, 2022-10-27:
Actually many GPS receiver deduce the speed using Doppler measurement, which give them the speed independently of the (previous/current/next) position.

I will take a moment to think about it

stefaniapedrazzi, 2023-01-11:
Addressed in #5372.

closed 2023-01-11 by stefaniapedrazzi