cyberbotics/webots #5373: cannot import sumo supervisor

karellyrl, 2022-10-26:
hi, i'm using webots ros2 and i added in a webots world the PROTO Sumo interface, but when i start the simulation i get this error:
```
Traceback (most recent call last):
  File "sumo_supervisor.py", line 17, in <module>
    from SumoSupervisor import SumoSupervisor
  File "/home/lab/webots/projects/default/controllers/sumo_supervisor/SumoSupervisor.py", line 17, in <module>
    from controller import Supervisor, Node
  File "/home/lab/webots/lib/controller/python38/controller.py", line 31, in <module>
    import _controller
ImportError: /home/lab/webots/lib/controller/python38/_controller.so: undefined symbol: _ZNK6webots5Robot7getTypeEv
```
I already exported this:
```
export WEBOTS_HOME=/home/lab/webots
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$WEBOTS_HOME/projects/default/resources/sumo/bin:$WEBOTS_HOME/lib/controller
export SUMO_HOME=/usr/share/sumo
export PYTHONIOENCODING=UTF-8
```
I also established the sumo interface in .wbt file as follows:
```
SumoInterface {
  gui FALSE
  useNetconvert FALSE
  radius 250
  laneChangeDelay 4
  networkFiles "/home/lab/ros2_ws/src/car_mdp/car_mdp/worlds/myMap_net"
}
```
 I read that the problem solves by exporting LD_LIBRARY_PATH, but it didn't work to me. I'm using ROS2 foxy, WebotsR2022a and Python 3.8.10. 

Thanks!

stefaniapedrazzi, 2022-10-27:
This might be due to a known issue with Supervisor and ROS2 solved after the release of Webots R2022a.

I would suggest you to upgrade to Webots R2022b and compile the `webots_ros2` module from sources:
https://github.com/cyberbotics/webots_ros2/wiki/Complete-Installation-Guide

karellyrl, 2022-10-28:
> This might be due to a known issue with Supervisor and ROS2 solved after the release of Webots R2022a.
> 
> I would suggest you to upgrade to Webots R2022b and compile the `webots_ros2` module from sources: https://github.com/cyberbotics/webots_ros2/wiki/Complete-Installation-Guide
Hi, i upgrade webots but now it doesn't start the simulation, it just keep in: [driver-4] Cannot open file: /tmp/webots-253336-2vJJ8A/WEBOTS_SERVER (retry count 15), and never initialize

stefaniapedrazzi, 2022-10-31:
WEBOTS_SERVER has been removes in R2022b.
So if you are getting this error it is very likely that your controller is still compiled against Webots R2022a  and not R2022b.
Did you also upgrade the `webots_ros2` module and re-build it other than installing the new version of Webots?
Are you sure that `webots_ros2` is using the Webots R2022b?

karellyrl, 2022-10-31:
> WEBOTS_SERVER has been removed in R2022b. So if you are getting this error it is very likely that your controller is still compiled against Webots R2022a and not R2022b. Did you also upgrade the `webots_ros2` module and re-build it other than installing the new version of Webots? Are you sure that `webots_ros2` is using the Webots R2022b?

I upgraded webots_ros2 and now the sumo interface works properly, thank u!

closed 2022-10-31 by karellyrl