cyberbotics/webots #5425: CSS for a robot window is not always served correctly

amrytov, 2022-11-05:
There was an HTML robot window (copied and modified from the "simple window" example) consisting of the HTML itself (some_window.html) and a CSS file (some_window.css), referenced by a <link> tag.  Sometimes the window opened normally (with styles), sometimes it failed to use the stylesheet (appeared plain HTML, black on white).   

Studying the sources in Chrome's dev panel shows that WeBots' local HTTP server correctly served the HTML file, then served _the same HTML file_ again instead of serving the CSS.  When I rename the CSS (and the URL in the <link> tag) and refresh the page in the browser, then the first time it's rendered correctly, but later the problem keeps happening.   (I had to circumvent the problem by putting the CSS directly into the HTML code and getting rid of a separate CSS source).

Looks maybe like incorrect caching in WeBots HTTP server. 

**System**
 - Windows 10

BenjaminDeleze, 2022-11-08:
Hi,
I tried to reproduce your issue, but did not manage to do so.

What version of Webots are you using?
How do you open your robot window?
How did you include your `css` file in the `.html`?

amrytov, 2022-11-09:
Version:  the "stable"  2022b from you Cyberbotics main site
System: Windows 10 with Chrome as the default browser.

How do I open the window:   in different ways. Sometimes from the context menu on the object, sometimes it like pops-up all by itself, unexpectedly (which by itself might be worth a bug report or a feature request).  I have not noticed whether or not the effect depends on the way the browser opens, and now that I got rid of the separate CSS file, I can't reproduce the effect easily.

How did I include your css file in the .html:  exactly the same way as it was done in the "simple robot window" example, via the *link* tag.  Actually copied the sample, renamed both HTML and CSS, changed URL in the *link* (it was a short relative one), changed to my needs the HTML in the *body* section.

stefaniapedrazzi, 2022-11-23:
Not sure if it is related.
I also have issues with the load of CSS files for robot windows when when using a `https://raw.githubusercontent.com/cyberbotics/webots/` URL (for example https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/samples/robotbenchmark/include/tabs.css): even if the URL is correct and the file can be open in the browser, when Webots opens the robot window the CSS is not applied (both locally and on the webots.cloud).
Note that if I save the robot window HTML page and reopen it in the browser, then it works correctly.

I could fix my issue by copying the CSS file locally instead of using the https://raw.githubusercontent.com/cyberbotics/webots/` URL.

BenjaminDeleze, 2023-06-06:
Closing inactive issue.

closed 2023-06-06 by BenjaminDeleze