cyberbotics/webots #5437: Webots R2022b won't load after code error

zackzack38, 2022-11-08:
**Describe the Bug**
The Webots program does not simulate after a code error with exit status 1. Previous version of Webots would show the error in the console, allow you to edit the code, and run successfully with the fixed code.

**Steps to Reproduce**
1. Open any world
2. Change the python code so that it has any syntax error
3. Go to run the simulation
4. No movement of the simulation despite "Starting controller" in console
5. See error

**Expected behavior**
The simulation would load and run with the fixed code.

**System**
 - Operating System: Windows 10
 - Graphics Card: AMD Radeon RX Vega 10 and NVIDEIA GeForce GTX 1660 Ti with Max-Q Design

ad-daniel, 2022-11-08:
Hi, this is a known issue and should be fixed already. Try installing Webots from the latest nightly build: https://github.com/cyberbotics/webots/releases/tag/nightly_4_11_2022

amrytov, 2022-11-09:
Ah, after having to restart WeBots yet another time, I was just going to report this bug.  Good thing it is already fixed, unfortunately the recent nightly build doesn't work, which I reported separately.

ygoumaz, 2022-11-11:
I am closing this issue for now, as it is probably fixed already. Feel free to reopen if needed.