cyberbotics/webots #5452: Long non-ASCII robot name crashes the controller.

amrytov, 2022-11-09:
**Describe the Bug**

See the subj.

**Steps to Reproduce**

1. Create an empty world
2. Put a Robot node in it. No child nodes are needed.
3. Leave the <generic> controller as is.
4. Name the robot with any long non-ascii name ("ЭтоДлинноеИмяДляРобота")
5. Run the simulation

** What happens:

I've just seen each of the following, in different combinations:

- the controller crashes (both mine and the <generic>, but this is not very reproducable, so no screenshot)
- errors starting the controller (see the screenshot)
- errors resetting the simulation (see the screenshot)

**Expected behavior**

Either accept the name as valid and then work correctly, 
or report that the name is too long or contains illegal characters and not accept it.

Suggested possible solution(s):  

(a) Check validity of the string assigned to the 'name' field (don't allow such names)
(b) don't use robot name directly to create pipe names or whatever, instead create some fixed-length harmless hash string from them,
so that any strings can be safely used.

**Screenshots**

[image]
[image]
[image]

**System**
 - Operating System: Windows 10

brettle, 2024-08-23:
I just ran into this bug when running the test suite on a Mac where my username was 11 characters long.
The servername passed to `QLocalServer` (on Unix) has the form  "/tmp/webots/_username_/1234/ipc/_encodedRobotNname_/intern" which is 29+usernameLength+encodedRobotNameLength. PR #5959 hashes encodedRobotNameLengths > 70 chars, but since the total length needs to be at most [104 chars](https://www.unix.com/man-page/mojave/4/unix/) (not 108 chars), that means the problem will still occur for users with usernames longer than (104-70-29) 5 characters. Since we have a total of 104-29=75 chars to work with, perhaps if either the username or the encodedRobotName is longer than 36 characters, it should get replaced with the first 36 characters of its hex hash.

brettle, 2024-08-23:
It's actually a bit more complicated than I described, because in a Snap environment the prefix include $(HOME) which can be arbitrarily large. I'll be submitting a fix shortly. It will leave the prefix portion (everything through "ipc/" unchanged and just (if necessary) use a hash of the encodedRobotName that has been truncated as needed to make the resulting servername string fit on even the most restrictive system (which on the [Linux man page](https://man7.org/linux/man-pages/man7/unix.7.html) says means keeping it under 92 chars).

closed 2024-09-13 by brettle