cyberbotics/webots #5462: Physics plugin: webots_physics_collide called on empty group

BenjaminDeleze, 2022-11-11:
I recently worked on a R2020b simulation with a physics plugin and found out that the number of time `webots_physics_collide` was called was not the same in R2020b and R2022b.

For the same (quite complex) world, R2022b had something around 600 additional calls to `webots_physics_collide` per time step out of 5000.

I made a minimal example to reproduce this: [physics.zip](https://github.com/cyberbotics/webots/files/9988931/physics.zip)

It will print in the terminal the number of time the `webots_physics_collide` is called at each time step.

In webots R2022b, it is 3. if you remove the *empty* group from the inner solid, the number of call will drop to 1.
In webots R2020b it is always 1.

The change happened in R2021a, maybe in [this PR](https://github.com/cyberbotics/webots/pull/2781).

I don't know enough about physic plugin to be able to answer to the following question:
- is it normal to call `webots_physics_collide` additional times when there is an empty group? it seems strange to me as I don't see how an empty group can have a collision.
- is it possible that this additional calls slow Webots down a lot (around 1/3) in the large example I mention at the beginning?

In conclusion, do you know if it is a bug or if it is the normal behavior?

omichel, 2022-11-11:
I don't know either. I agree it seems strange to call the `webots_physics_collide` when there is an empty group. So, I think it is a bug...

BenjaminDeleze, 2022-11-21:
I investigated a bit more and think that this is a side-effect of #2781.

A possibility to avoid collisions with an empty group in the physics plugin would be to add: 
```
  if (dGeomIsSpace(o1)) {
    const dSpaceID spaceId = WbSolidUtilities::dynamicCastInSpaceID(o1);
    if (spaceId && dSpaceGetNumGeoms(spaceId) == 0)
      return;
  }

  if (dGeomIsSpace(o2)) {
    const dSpaceID spaceId = WbSolidUtilities::dynamicCastInSpaceID(o2);
    if (spaceId && dSpaceGetNumGeoms(spaceId) == 0)
      return;
  }
```
At the beginning of `WbSimulationCluster::odeNearCallback`.

However, it was decided after some discussions with @omichel that we will not implement that. We do not want to add some additional check at each time step and for every collision, moreover having an empty group in bounding object is a bad practice and should be avoided.

closed 2022-11-21 by BenjaminDeleze