cyberbotics/webots #5470: Linker Error wb_robot_wwi_receive_text not in Controller.lib

Leo-Nogismus, 2022-11-11:
**Describe the Bug**
I'm trying to build a controller for WeBots using Visual Studio 2019 Enterprise. When building the solution I get Linker Error "Robot.obj : error LNK2001: unresolved external symbol wb_robot_wwi_receive_text". All other external symbols link correctly. When opening Controller.lib in Notepad I can find all other external symbols, but not "wb_robot_wwi_receive_text".

**Steps to Reproduce**
1. Create a new controller using the Webots File > New > New Robot Controller
2. When prompted select "Microsoft Visual Studio"
3. Follow the instructions for linking, include libs, etc on https://www.cyberbotics.com/doc/guide/using-visual-studio-with-webots?version=cyberbotics:R2019a (More or less all of it is set correctly already)
4. Try to build the solution.

**Expected behavior**
No linker errors should occur

**System**
 - Operating System: Windows 10
 - Visual Studio 2019 Enterprise

Leo-Nogismus, 2022-11-11:
Tried to build the same project using WeBots 2022a instead of WeBots 2022b and it works. I guess it is a bug in connection with commit efeefd05f3b41da8e9a68fbd6508048ef660fd8f or commit efeefd05f3b41da8e9a68fbd6508048ef660fd8f not sure which one.

omichel, 2022-11-11:
I just found and fixed the problem. If this PR gets approved and merged today, you will be able to test the nightly build of Webots R2023a tomorrow including this patch.

closed 2022-11-11 by ad-daniel

Leo-Nogismus, 2022-11-11:
That was quick, thank you very much! As it is already merged I will test it tomorrow! Thanks a lot!

PR #5471 (2022-11-11): Fixed missing exported function for Visual C++
Fixes #5470.