cyberbotics/webots #5477: Cannot open any world file on macOS with R2023a nightly build

omichel, 2022-11-14:
I am getting this error in the console:
```
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/backgrounds/protos/TexturedBackground.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/backgrounds/protos/TexturedBackgroundLight.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/trees/protos/Sassafras.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/trees/protos/Sassafras.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/vehicles/protos/abstract/VehicleWheel.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/robots/clearpath/moose/protos/MooseWheel.proto': error: PROTO has unknown base type.
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/vehicles/protos/abstract/VehicleWheel.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/robots/clearpath/moose/protos/MooseWheel.proto': error: PROTO has unknown base type.
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/vehicles/protos/abstract/VehicleWheel.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/robots/clearpath/moose/protos/MooseWheel.proto': error: PROTO has unknown base type.
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/vehicles/protos/abstract/VehicleWheel.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/robots/clearpath/moose/protos/MooseWheel.proto': error: PROTO has unknown base type.
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/vehicles/protos/abstract/VehicleWheel.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/robots/clearpath/moose/protos/MooseWheel.proto': error: PROTO has unknown base type.
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/vehicles/protos/abstract/VehicleWheel.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/robots/clearpath/moose/protos/MooseWheel.proto': error: PROTO has unknown base type.
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/vehicles/protos/abstract/VehicleWheel.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/robots/clearpath/moose/protos/MooseWheel.proto': error: PROTO has unknown base type.
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/vehicles/protos/abstract/VehicleWheel.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/robots/clearpath/moose/protos/MooseWheel.proto': error: PROTO has unknown base type.
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/factory/containers/protos/CardboardBox.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/factory/containers/protos/CardboardBox.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/factory/containers/protos/CardboardBox.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/factory/containers/protos/CardboardBox.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/factory/containers/protos/CardboardBox.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/trees/protos/Sassafras.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/trees/protos/Sassafras.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/trees/protos/Sassafras.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/trees/protos/Oak.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/trees/protos/Oak.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/trees/protos/Oak.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/trees/protos/Oak.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/trees/protos/Oak.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/trees/protos/Oak.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/trees/protos/Oak.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/trees/protos/Oak.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: 'https://raw.githubusercontent.com/cyberbotics/webots/028694df3179b81396ee615707832aec654e0161/projects/objects/trees/protos/Oak.proto': error: Template engine error: Couldn't write jsTemplateFilled to disk..
ERROR: Cannot listen to the local server (server name = '/var/tmp/webots-1234/ipc/moose/intern'): QLocalServer::listen: Name error
```

omichel, 2022-11-14:
This problem was due to the fact that another user had run Webots before and it crashed Webots, so that the `/tmp` file of Webots was not cleaned and could be cleaned by the current user.
To fix the problem I had to:
```bash
sudo rm -rf /private/var/tmp/webots-1234
```
To fix this problem, we should use a local tmp folder instead of a global one.

ad-daniel, 2022-11-15:
We can probably close this now, has it been tested with the latest nightly?

omichel, 2022-11-15:
Yes, it works now.

PR #5484 (2022-11-14): Fixed tmp path on macOS
Fixes #5477.

This PR makes that Webots uses the local user tmp folder instead of the global one, thus avoiding problems like the one raised in #5477.
closed 2022-11-15 by omichel