cyberbotics/webots #5479: External force and torque commands don't require the `ctrl` button as stated

ad-daniel, 2022-11-14:
**Describe the Bug**

According to the help menu, to extert a torque on an object you have to (applies to forces too):
> Place the mouse pointer on the object and hold down the Alt key, the Control key (Ctrl) and the right mouse button together while dragging the mouse.

In practice it appears to be sufficient to press the Alt, Ctrl is not needed. Same thing for the force.

Tested on snap, under ubuntu 20.

ad-daniel, 2022-11-14:
So it seems that for forces the options are:
- alt + left mouse
- ctrl + alt + left mouse

for torque:
- alt + right mouse
- ctrl + alt + right mouse

This is the behavior for linux. On mac it appears to be different (ctrl + alt + left makes a torque, not a force)

omichel, 2022-11-14:
I believe that we should document only:

If your mouse has two buttons (e.g., on Linux, Windows and some Mac):
- To apply a force: <kbd>alt</kbd> + left mouse drag.
- To apply a torque: <kbd>alt</kbd> + right mouse drag.

If your mouse only has a single button (e.g, on Mac):
- To apply a force: <kbd>alt</kbd> + mouse drag.
- To apply a torque <kbd>alt</kbd>+<kbd>ctrl</kbd> + mouse drag.

closed 2022-11-14 by ad-daniel