cyberbotics/webots #5491: Motion blur not applied on pasted robot with camera (Windows)

stefaniapedrazzi, 2022-11-15:
Steps to reproduce:
1. Open `camera_motion_blur.wbt` and run the simulaton
2. Copy the MyBot robot and paste it in the world
-> in the camera overlay of the newly added Robot the motion blur effect is not visible

I get this behavior on Windows (with user name containing non UTF-8 characters).
I tested on my development system on Ubuntu 22.04 and the motion blur is applied correctly to the pasted Camera image.

omichel, 2022-11-15:
I am getting the same bug on macOS although my user name is fully ASCII.

omichel, 2022-11-15:
I also noticed on macOS that if the ambient occlusion is disabled in the Webots/OpenGL preferences, the motion blur doesn't work at all even for the first robot.

stefaniapedrazzi, 2022-11-16:
It is not reproducible, but sometimes also on my Windows machine it doesn't work even for the first robot.

omichel, 2022-11-16:
Yes, it is the same also on macOS.

stefaniapedrazzi, 2022-11-17:
I can reproduce this issue with the nightly built before changind the GTAO shader.

PR #5517 (2022-11-17): Fix motion blur shader
Possibly fix #5491 and #5515.
"firstRender" uniform is created with the wrong type (introduced in #5436).
This might cause the not fully reproducible issues experienced on various systems.
closed 2022-11-23 by stefaniapedrazzi