cyberbotics/webots #5492: Laser visualization update rate differs froms distance sensor rays

stefaniapedrazzi, 2022-11-15:
In `laser_pointer.wbt`, the laser point visualization update rate (64 ms) is lower than the update rate of the distance sensor rays (optional rendering updated at basicTimeStep 8ms).
This is not really good because most of the time the laser pointer are noticeable not at the correct position and not aligned with the DistanceSensor rays.

It would be better to increase use the basicTimeStep as sampling rate for the DistanceSensor device so that the visualization of the laser point and sensor rays is always consistent.

[image]

ShuffleWire, 2022-11-15:
AFAIK, it remind me https://github.com/cyberbotics/webots/issues/3168 . I'm correct. It should be a good motivation to fix it together

ShuffleWire, 2022-11-15:
On a broader scale, their is **A LOT** of `#define TIMESTEP 32` thingies in the controllers codebase. (sample, example, tests...), and I feels it should be good to get ride of it altogether...

omichel, 2022-11-15:
Sure, but let's do it step-by-step, this is more manageable.

stefaniapedrazzi, 2022-11-16:
There are some good reason to use a different time step in the controller than the world basic time step.
Unless this is causing issues, like in this example, we should not change all of them.

ShuffleWire, 2022-11-17:
> There are some good reason to use a different time step in the controller than the world basic time step. Unless this is causing issues, like in this example, we should not change all of them.

What example do you have in mind ? Just to be able what to change or not before hitting the review.... Beside case when we specifically want to test time_step behavior, ofc

stefaniapedrazzi, 2022-11-17:
The choice of the controller time step and basic time step is very often made so that to improve the physics computation precision without slow down too much the simulation.
So, unless it is causing particular issues we should not change the controller step of tests or any other examples in batch.

Modifying the tutorials controllers and the default controllers is the only improvement that seems to be meaningful.

commit 045000f6 (2023-01-31): Laser visualization update rate differs from distance sensor rays #5492 (#5801)

* Resolve #5492: Updated controller time step - laser_pointer.c

* Resolve #5492: Updated controller time step to robot basic time step - laser_pointer.c

* Resolve #5492: Updated controller time step - laser_pointer.c

* Resolve #5492: Clang format correction

---------
closed 2023-01-31 by omichel