cyberbotics/webots #5493: Loading dialog continuously open and closes while loading world

stefaniapedrazzi, 2022-11-15:
On Windows, it happens that when opening a world the loading dialog continuously disappears and re-appears when the following text is displayed "Open World File".
This blinking effect is a little bit disturbing and should be avoided.

I could reproduce this on Windows (didn't check on Ubuntu).

omichel, 2022-11-15:
I confirm this problem also shows up on macOS.

ad-daniel, 2022-11-15:
This might actually be a regression, it doesn't happen with the release of R2022b

ad-daniel, 2022-11-16:
The problem appears to stem from 2 conflicting progress counters:
- [This](https://github.com/cyberbotics/webots/blob/48444fa1f9d6213ef6efc0f69f63dbbd13db692f/src/webots/app/WbApplication.cpp#L53-L54) check says the progress is 100 and therefore the dialog is closed
- meanwhile, occasionally, [this](https://github.com/cyberbotics/webots/blob/48444fa1f9d6213ef6efc0f69f63dbbd13db692f/src/webots/app/WbApplication.cpp#L142-L147) setter re-creates the dialog because according to the parameter it receives, the progress is not 100

basically the two fight with each other

omichel, 2022-11-18:
I believe we can close this issue now.

closed 2022-11-18 by omichel