cyberbotics/webots #5515: Motion blur test is broken

ad-daniel, 2022-11-16:
**Describe the Bug**

As emerged in https://github.com/cyberbotics/webots/pull/5505, the motion_blur test occasionally fails but was until now silently ignored (didn't trigger the test suite to fail in the CI). I'm having a hard time trying to reproduce it locally, but I see it appear now and then by:

1. open `camera_color_motion_blur.wbt`
2. set fast mode
3. keep reloading over and over, might take 10-20 tries before it shows, but it shows

stefaniapedrazzi, 2022-11-16:
Might be related to #5491

ad-daniel, 2022-11-16:
Unfortunately it turns out to be more recent, it seems this GTAO shaders fix https://github.com/cyberbotics/webots/pull/5504 introduces the problem

closed 2022-11-18 by ad-daniel