cyberbotics/webots #5522: The simulation is slow.

vitormartins01, 2022-11-19:
My simulation has 1 dynamic robot and a camera. The problem with the simulation getting stuck is when I activate the camera controller...
I'm using the latest version of webots R2023A, I've tried everything to try to solve this problem but I couldn't find a solution.

System 
- System: Windows 11
 - Graphics Card:

RugnirViking, 2022-11-19:
what graphics card do you have? and is the problem that the simulation is slow or that it gets "stuck"?

vitormartins01, 2022-11-19:
Graphics Card: Nvidia Geforce gtx 1650 ti.

I didn't understand your question !

Without the camera controller activated my robot moves normally (at the speed I set) and the simulation is running at 0.9X. With the camera controller activated the robot appears stuck and the simulation is running at 0.01x.

vitormartins01, 2022-11-20:
Could someone help me ?

stefaniapedrazzi, 2022-11-21:
This issue seems to be duplicated in #5524.

closed 2022-11-21 by stefaniapedrazzi