cyberbotics/webots #5524: The simulation is stucked.

vitormartins01, 2022-11-20:
My simulation has 1 dynamic robot and a camera. The problem with the simulation getting stuck is when I activate the camera controller with the robot controller activated...
I'm using the latest version of webots R2023A, I've tried everything to try to solve this problem but I couldn't find a solution.

System

System: Windows 11
Graphics Card: Nvidia Geforce gtx 1650 ti.

omichel, 2022-11-21:
Could you please share the source code of a very simple simulation showing this problem?

vitormartins01, 2022-11-21:
Yes i can, only have one robot with a camera and an "e-puck robot".
[TESTE_2023A (2).zip](https://github.com/cyberbotics/webots/files/10055739/TESTE_2023A.2.zip)

omichel, 2022-11-21:
I tested your world file name `my_world.wbt` and it worked without being stuck.
I just noticed that it is very slow, but this is likely due to your controller doing heavy image processing on a high resolution camera image.

vitormartins01, 2022-11-21:
I use that resolution because i need it for a good image processing.

The simulation didnt stop but was very slow.

omichel, 2022-11-21:
Then, this is not a bug. It is due to the complexity of your controller program which is where is the bottleneck.
I would recommend to try to rewrite your program in C or C++ instead of Python.
It should run faster.
Copying large images in Python is known to be very slow.

closed 2022-11-21 by omichel