cyberbotics/webots #5540: RpLidarA2 - rotating bugs meansurements

delihus, 2022-11-21:
**Describe the Bug**
While preparing package `webots_ros2_husarion` I discovered that RpLidarA2 is not working properly.
[image]

[image]

[Here](https://github.com/cyberbotics/webots/blob/develop/projects/devices/slamtec/protos/RpLidarA2.proto) commented this line:
```
# type "rotating"
```
[image]

ygoumaz, 2022-11-22:
I can't reproduce your issue. In the sample world the Lidar seems to work perfectly fine.

[image]

Can the problem come from the ROS2 package? Or maybe from RViz? Are you making any transformation to the Lidar data before displaying it?

Could you please provide a minimal example with the failure, so that I can reproduce it?

ygoumaz, 2022-12-13:
After further investigation on the Lidar and your project (following up our discussion on Discord), there is indeed a bug with the (rotating?) Lidars. Here is what I have observed:

- All Lidars display correctly the point cloud inside Webots, including `RPLidarA2`, whatever their field of view is.
- As reported by @delihus, the measurements of the Lidar (not the point cloud) are wrong for a field of view of 1.57 on the RPLidarA2. This corresponds to the `/scan` topic in `webots_ros2` and has only been observed in RViz (for now).
- Increasing the field of view to 6.28 fixes the `RPLidarA2` measurements.
- The point cloud is not correctly displayed in Rviz (`/scan/point_cloud` topic). Points are correct but the complete pattern is rotated by 180 degrees.

I still need to investigate if the wrong measurements are coming from Webots or when writing to `webots_ros2` topics.

delihus, 2022-12-13:
Thank you @ygoumaz!

I will also show the others what is happening  then [slam-toolbox](https://github.com/SteveMacenski/slam_toolbox) is mapping.

[image]
[image]

I can't wait. 
Best regards,
Jakub

toshiharutf, 2023-01-05:
Hello,

I faced the same issues mentioned using slam_toolbox with Webots. I observed the odometry values coming from the diff_drive node started to drift quite a lot in short time period. As mentioned in the documentation, slam_toolbox requires high quality odometry, so this could be one of the reasons. Also, fine tuning of the config params may be required

I got better results using cartographer out of the box with the turtlebot demo: https://ros2-industrial-workshop.readthedocs.io/en/latest/_source/navigation/ROS2-Cartographer.html. My guess is that the algorithm in cartographer is more robust to noisier odometry.

I hope this could help.

Best regards,

delihus, 2025-03-10:
@shiven2001 
Actually I used cartographer_ros instead of slam_toolbox and it does not have this issue