cyberbotics/webots #5642: BotStudio not opening

felipenmartins, 2022-12-07:
**Describe the Bug**
The graphical robot window "BotStudio" does not open.

**Steps to Reproduce**
1. Click on File > Open Sample Worlds and go to robots > gctronic > e-puck 
2. select e-puck_botstudio_with_floor_sensors.wbt
3. Doubleclick on the e-puck robot

**Expected behavior**
The BotStudio window should open.

**System**
 - Operating System: Windows 10
 - Webots R2023a
 - Graphics Card: Intel UHD Graphics 620

**Additional context**
It used to work well on Webots 2021b.

In R2023a I also tried to open BotStudio by right-clicking on the robot and selecting "Show Robot Window" option. I got the following messages on the Webots console:

Error: The specified module could not be found.
 (dynamic library)
Error: robot window initialization failed
Error: Cannot load the "C:/Users/nasf/AppData/Local/Programs/Webots/projects/robots/gctronic/e-puck/plugins/robot_windows/botstudio/botstudio.dll" robot window library.

stefaniapedrazzi, 2022-12-08:
BotStudio is an old robot window based on Qt.
Since Webots R2022b only HTML based robot windows are supported, so BotStudio is no longer working on Webots R2023a.
Unfortunately, it was not removed from the Webots package when it stopped to work but we will remove it for the next version.

We do not plan to port it to HTML, but if you are willing to contribute we will be happy to review your contribution or give your some tips about how to proceed for implementing BotStudio with the new robot window structure (see [documentation](https://www.cyberbotics.com/doc/guide/controller-plugin#robot-window)).

stefaniapedrazzi, 2022-12-08:
Even if it is no longer officially support you can still make the BotStudio window working by adding a `runtime.ini` file in the controller folder similar to this one:
```
; we need to include the path to the Qt libraries because of the native Qt robot window used by this robot
[environment variables for Linux]
WEBOTS_LIBRARY_PATH = $(WEBOTS_HOME)/projects/robots/gctronic/e-puck/plugins/robot_windows/botstudio:$(WEBOTS_HOME)/resources/projects/libraries/qt_utils:$(WEBOTS_HOME)/resources/projects/libraries/qt_utils:$(WEBOTS_HOME)/lib/webots
[environment variables for macOS]
WEBOTS_LIBRARY_PATH = $(WEBOTS_HOME)/projects/robots/gctronic/e-puck/plugins/robot_windows/botstudio:$(WEBOTS_HOME)/Contents/Resources/projects/libraries/qt_utils:$(WEBOTS_HOME)/Contents/lib/webots
[environment variables for Windows]
WEBOTS_LIBRARY_PATH = "$(WEBOTS_HOME)\projects\robots\gctronic\e-puck\plugins\robot_windows\botstudio":"$(WEBOTS_HOME)\resources\projects\libraries\qt_utils;$(WEBOTS_HOME)\msys64\mingw64\bin"
```

Here is the `runtime.ini` file documentation:
https://www.cyberbotics.com/doc/guide/controller-programming#environment-variables

felipenmartins, 2022-12-12:
> I just added the required runtime.ini file to the repo (and to the Webots
> packages) so that in the next version it should work out of the box.

Thanks a lot! :-)