cyberbotics/webots #5690: VR view dose not show the environment on Webot after 2019a version.

AliPakbaz98, 2022-12-22:
Hello everyone, I have a problem for VR viewpoint on webot on all version after 2019a. 
I have tried both HTC VIVE and OCULUS to see VR view after that version, but unfortunately I see a black screen.

Problem: 
When I change view to VR view there is only black screen and if I click on an object I can see XYZ arrow of that object on black screen. I also putted human body and enabled showing joint axis and in VR view, it just shows joint axis. 

I think TexturedBackground dose not work on VR view and my view on VR headset is like when you turn off the Texturedbackground on screen view!

I have attached the headset view to this chat.

Can you help me how to fix this problem to see the environment in the VR? because I need to work with 2023 version.

[image]

omichel, 2022-12-23:
Duplicate with #747.

omichel, 2022-12-23:
**Note:** please write to sales@cyberbotics.com if you want to get a quote from Cyberbotics to fix this issue.

closed 2022-12-23 by omichel