cyberbotics/webots #5708: Position mismatch with velocity control of a RotationalMotor on a Hinge2Joint

CoolSpy3, 2023-01-04:
**Describe the Bug**
When using velocity control with a RotationalMotor attached to a Hinge2Joint (or HingeJoint), there accumulates a mismatch between the position of the joint in the world tree and the position of the joint in the simulation.

**Steps to Reproduce**
1) Open the sample world `samples > devices> hinge_2_joint_with_backlash.wbt`
2) REPLACE line 46 in the controller with:
```
wb_motor_set_position(motor1, INFINITY);
wb_motor_set_velocity(motor1, fabs(cos(0.01 * counter)));
```
3) Save and build the controller, and reload the world
4) Let the world run in fast mode for a while (I waited until an effective time of 24 minutes had elapsed)
5) Pause the world
6) In the world tree, open `DEF WIHTOUT_BACKLASH Robot > children > Solid "solid" > children > Hinge2Joint > JointParameters HingeJointParameters > position`
7) Without changing the value, click in the value entry box and press ENTER
8) The joint moves to a significantly different position

With a HingeJoint, the behavior is a little harder to see visually, but I've been able to reproduce it with:
1) Open the sample world `samples > devices> hinge_joint_with_backlash.wbt`
2) Let the world run in fast mode for a while (I waited until an effective time of 24 minutes had elapsed)
3) Pause the world
4) In the world tree, open `Robot "robot" > children > HingeJointWithBacklash > JointParameters HingeJointParameters > position`
5) Without changing the value, click in the value entry box and press ENTER
6) The joint moves to a significantly different position

**Expected behavior**
There should be no change after forcing an update to the position in the world tree because it should already be the value in the simulation.

**Screenshots**
Hinge2Joint after running simulation:
[image]

Hinge2Joint after running simulation and forcing position update to world tree
[image]

HingeJoint after running simulation
[image]

HingeJoint after running simulation and forcing position update to world tree (look at the blue flag)
[image]

**System**
 - Operating System: Windows 10
 - Graphics Card: NVIDIA GeForce GTX 1050 Ti
 - Webots: 2023a

**Additional context**
N/A

ad-daniel, 2023-01-04:
At first glance I thought it might be related to an accumulation of error due to the CFM/ERP constraints imposed in order to simulate the backlash, but it seems that doing the same change in the `motor2.wbt` world where there is no backlash involved, the same happens.

By ensuring we switch from position to velocity control only once (rather than every timestep) with
```
if (counter == 500)
  wb_motor_set_position(motor1, INFINITY);
```
the problem seems to go away. Still doesn't explain why it accumulates the error, but might give a clue

CoolSpy3, 2023-01-04:
I'm not able to reproduce this fix on my end :/

Following the steps above to load `hinge_2_joint_with_backlash.wbt`, replacing line 47 with `wb_motor_set_velocity(motor1, fabs(cos(0.01 * counter)));`, and adding `if (counter == 500) wb_motor_set_position(motor1, INFINITY);` after line 35 still leads to an accumulation of error.

I should also note that I've been able to confirm that this also occurs on the orthogonal axis of the Hinge2Joint (follow the steps above and replace `motor1` with `motor2`).

CoolSpy3, 2023-01-04:
I've done some experimenting across versions, and I've been able to reproduce it as far back as R2020b-rev-1 which was the first version of Webots I used. I think it's safe to say this rules out any recent changes to the simulator. Later I might look at older versions and try to figure out which version introduced the bug.

brettle, 2023-01-05:
I suspect the problem lies [here](https://github.com/cyberbotics/webots/blob/R2023a/src/webots/nodes/WbHinge2Joint.cpp#L432) (and similar logic for [the 2nd axis of the `Hinge2Joint`](https://github.com/cyberbotics/webots/blob/1c1d0fe4c5c4fd1519d772841e40df9905811a8e/src/webots/nodes/WbHinge2Joint.cpp#L444) and [the only axis of the `HingeJoint`](https://github.com/cyberbotics/webots/blob/1c1d0fe4c5c4fd1519d772841e40df9905811a8e/src/webots/nodes/WbHingeJoint.cpp#L383). Basically, Webots computes the position differently when the rotational motor is doing velocity control. Instead of using the actual position of the joint, it uses the previous position of the the joint and the post-physics velocity of the joint to calculate the new position. 

This was done intentionally "because at high speed angle feedback is under-estimated". I'm not sure exactly what that refers to, but my guess is that this is because the ODE physics engine always returns a joint angle between -pi and pi and Webots wants to report a position that doesn't have that constraint (eg if a joint that has done exactly 2 revolutions, ODE would report an position of 0 but Webots wants to report a position of 4\*pi). To accomplish that, Webots keeps track of the non-normalized position and uses the results from ODE to update that position each timestep. If the joint speed is higher than pi/timestep, the normalized angle reported by ODE is not sufficient to correctly update the non-normalized position. For example, if the joint speed is exactly 2\*pi/timestep, the normalized angle reported by ODE would never change and so neither would the non-normalized position.

That all makes sense to me, but I think the fix for high speeds that is currently implemented causes the bug we are observing because it doesn't ensure that the position reported by Webots actually corresponds to the normalized angle reported by ODE. And the fact that the current fix for high speeds uses the post-physics velocity guarantees that it will be off by at least a small amount if the joint has accelerated during the time step. And those errors accumulate.

Imo, independent of velocity control vs position control, Webots should:

1. Use the joint velocity (ideally the average of the pre-physics and post-physics velocity) to do a "rough" update to the joint position, and then
2. Refine the joint position so that it actually corresponds to the normalized joint angle reported by ODE.

Thoughts?

omichel, 2023-01-05:
I believe you analysis makes sense and there is certainly room for improvement here. Please go ahead and propose a patch that would improve the accuracy as suggested and fix the problem.

omichel, 2023-01-05:
Yes, that's a hotfix and should target the master branch.

Regarding the `BallJoint`, this should probably be addressed in another PR aiming at the master branch as well (unless it is closely related with the `Hinge2Joint` bug).

omichel, 2023-02-09:
By the way, can we close this issue as it seems it was resolved in #5834?

brettle, 2023-02-09:
Yes. @CoolSpy3, this is fixed in the latest nightly release.

closed 2023-02-09 by brettle

CoolSpy3, 2023-02-09:
Can confirm; the issue is fixed! Thank you all for your help!