cyberbotics/webots #5757: calling wb_supervisor_node_get_field with proto internal field causes performance degradation

jgueldenstein, 2023-01-19:
**Describe the Bug**
The problem lies in `webots/src/controller/c/supervisor.c`
When calling `wb_supervisor_node_get_field` (https://github.com/cyberbotics/webots/blob/master/src/controller/c/supervisor.c#L2367) with a proto internal node as `WbNodeRef` the field list is searched (https://github.com/cyberbotics/webots/blob/master/src/controller/c/supervisor.c#L2384) with the parameter `is_proto_internal` set to false by default.
As the node is proto internal, this will always fail.
Therefore, the field is newly requested from the simulator and the first element of the field_list is returned.

However, this causes the field list to grow.
If this is done often, the field_list grows to an unmanageable size causing significant performance degradation.

This function may be called often, for example when getting contact points using `wb_supervisor_node_get_contact_points` with include_descendants = true, then getting the `WbNodeRef` using  `wb_supervisor_node_get_from_id` using the node_id in he contact point struct (https://cyberbotics.com/doc/reference/supervisor#contact-point) to get the `WbNodeRef`. When calling `wb_supervisor_node_get_field` on this reference (for example to get the name of the solid which made contact), the above described behavior occurs.

**Expected behavior**
performance should not degrade 

**System**
 - Operating System: Linux Ubuntu 22.04
 - should affect all systems

**Additional context**
I found a workaround which is implemented which is implemented here: https://github.com/RoboCup-Humanoid-TC/hlvs_webots/pull/7

jgueldenstein, 2023-01-19:
This problem was introduced by https://github.com/cyberbotics/webots/commit/d45037c97eeaf46622e2973259e4d0ec582c8567 and is not present in 2021b

stefaniapedrazzi, 2023-01-25:
This is fixed in https://github.com/cyberbotics/webots/pull/5774.
The patch will be included in the nightly build of Webots R2023a rev1 starting from tomorrow.

closed 2023-01-25 by stefaniapedrazzi

jgueldenstein, 2023-01-25:
Thank you so much for providing a fix

PR #5774 (2023-01-25): Fix retrieving cached PROTO internal fields in Supervisor API
Fix #5757: add a new flag to `WbFieldStruct` to mark if the field is a PROTO field or parameter.

In the Supervisor API the the `WbFieldStruct.is_proto_internal` flag was used to determine if the cached field was read only (i.e. not visible from the scene tree).
Then in #4363, it was added a check was added to make sure that if a PROTO field and parameter with the same name exist the correct one is returned. This check is fine and is needed to retrieve the correct field.
However, the value contained in the former `WbFieldStruct.is_p [... cut]