cyberbotics/webots #5768: Contact points (other than the first 10) are silently ignored

brettle, 2023-01-22:
**Describe the Bug**
When 2 geometries collide, only the first 10 contact points discovered are used to create contact joints. Others are silently ignored. This can lead to strange behavior. In particular, if 2 spatially separated regions of a geometry collide, the first 10 contact points found might be in one of those regions and as a result the other region will pass through.

**Steps to Reproduce**
1. Open ContactPointTest.wbt in the attached project. It contains a traffic cone mesh falling onto a slanted box.
2. Stop and reset the simulation if necessary.
3. Show bounding boxes and contact points if necessary.
4. Step through the simulation until the base of the traffic cone should be colliding with the box.
5. See how there are no contact points shown at the base of the traffic cone but there are many shown at the top of the cone. Note that there are no error or warning messages in the console associated with this behavior.

**Expected behavior**

The base of the traffic cone should also collide with the box. 

At a minimum, a warning message should appear in the console indicating that the maximum allowed number of contact points were found and additional contact points will be ignored. Ideally, all contact points would be used but I realize that is not practical. See below for ideas on strategies that ensure that the base of the traffic cone collides with the box even if not all contact points are used.

**Screenshot**
[image]

The problem code is [here](https://github.com/cyberbotics/webots/blob/52225cf830de223477da74dc35e0fe17fba48cf2/src/webots/engine/WbSimulationCluster.cpp#L631-L632).

Ideally, the other contact points would not be ignored. Unfortunately, since the computational complexity of a step in the ODE physics engine increases with the cube of the number of joints, it presumably isn't practical to create a contact joint for every contact point. However, instead of creating a contact joint for the first 10 contacts that happen to be found, I recommend  finding a much larger number of contact points and creating contact joints only for the 10 deepest. The idea here is that initially all of the contact joints will still be at the top of the cone, but they will limit the penetration of the top and eventually the base of the cone will penetrate further than those points causing joints to be created at the base that will supplant some of the joints at the top of the cone. I have a patch that implements this scheme and it does in fact solve the problem for this test case. My implementation finds the first 100 contact points and chooses the 10 deepest.

However it isn't clear that 100 contact points is the correct number to consider. For example, I can imagine a very detailed mesh where all 100 contact points are in the same region. To address the possibility of all of the contact points coming from the same region, I recommend randomizing the order that triangles are collided (using a shuffle weighted by the triangle area). This could presumably be done once when meshes are imported. The result would be stochastic spreading of the contact points around the mesh. The idea is that it would be extremely unlikely that of the first 100 contact points found, none of them would be in a particular region. I'm still working on implementing this. [Aside: This might also reduce the frequency of the "The current physics step could not be computed correctly" warning caused by ODE's LCP solver not converging.]

Thoughts?

Also, as I work on a fix for this, do you prefer the PR against master or develop?

[ContactPointTest.zip](https://github.com/cyberbotics/webots/files/10473395/ContactPointTest.zip)

stefaniapedrazzi, 2023-01-23:
Thank you for proposing this improvement!
I would say it is better to target the develop branch as the changes are likely to change simulation behaviors.

I see that in the PR you created you didn't implement randomizing the order that triangles are collided.
So, if you plan to add it we will make a full review and deep test of your PR when it is completed.

brettle, 2023-01-23:
Unfortunately, the randomizing I proposed is not effective because the collision algorithm does not process the triangles in the order they are listed in the mesh data structure. For performance reasons, it puts them in a spatial tree  and this results in nearby triangles being near each other in the evaluation order regardless of the order they are listed in the mesh. I may still investigate other options for ensuring that the contact points are spread out, but if I do I'll put it in a separate issue/PR.

ShuffleWire, 2023-01-31:
Dammed ! that kinda a big deal, tbf, and I will follows closely what you are doing, because I'm running in situation where I'm colliding complex shape, and got very bad result ! Idk if it's causing that (I expecting that it's not the only issue, but that seems plausible) 
Please document nicely your work, that other like me will not fall right into that trap later xD

closed 2023-02-09 by brettle