cyberbotics/webots #5780: Test `robot_parallel_step` failing on local machine running in fast mode

stefaniapedrazzi, 2023-01-25:
On my machine (Ubuntu 22.04) the `robot_parallel_step` api test is failing when it is ran in fast mode.
Running it at normal speed works correctly.

I experienced this on the develop branch (R2023b) compiled in debug mode.
I will also test on the master branch and with Webots compiled in release mode.

stefaniapedrazzi, 2023-01-26:
This is the error I get in the Webots console:
```
Warning: wb_robot_step() called before calling wb_robot_step_end().
Warning: robot_step_begin() called multiple times before calling wb_robot_step_end().
FAILURE with robot_parallel_step: Camera measurement after first pose update is wrong.
WARNING: 'robot_parallel_step' controller exited with status: 1.
```

ad-daniel, 2023-01-26:
tests in the CI should also run in fast mode, no?

stefaniapedrazzi, 2023-01-26:
Yes.
I'm now inspecting what occurs in my machine that makes the test fail on the develop branch only.

brettle, 2023-02-07:
Fwiw, the first 2 warnings are just a result of the assertion code calling `wb_robot_cleanup()` when an assertion fails. `wb_robot_cleanup()` calls `wb_robot_step()` which produces the warnings when a step is in progress. Of course, the real mystery is why assertion fails to begin with.

brettle, 2023-02-08:
I'm pretty confident that the issue is that images are being transferred from webots to the controller via a memory mapped file. That means that webots writes and controller reads of the image data could occur in an undefined order. When the error occurs, webots has updated the image before the controller has had a chance to read it and check it. I'm going to investigate using a double-buffering scheme to avoid this.

brettle, 2023-02-08:
Can I add a private member variable to [`WbAbstractCamera`](https://github.com/cyberbotics/webots/blob/master/src/webots/nodes/WbAbstractCamera.hpp) in a PR targeting master? I just need to avoid changing any wire protocols (e.g. the one between controllers and webots) or making ABI-incompatible changes to shared libs (e.g. `libController.so`), right? I obviously want to preserve compatibility on the master branch, but unless I'm missing something, those `src/webots/nodes/*` classes are only compiled into the webots executable. They aren't compiled into any shared lib where ABI compatibility would be an issue.

brettle, 2023-02-08:
Alternatively (and much more simply), we just document that the contents of image data is not well-defined between `wb_robot_step_begin()` and `wb_robot_step_end()`.

omichel, 2023-02-08:
Yes, adding a private member variable in a Webots class on the master branch is perfectly fine as long as you don't break the ABI compatibility between Webots and the Controller library, which shouldn't be the case.

brettle, 2023-02-08:
@stefaniapedrazzi, I'm going to start working on a PR against `released` for the documentation update. I'll also fix the test to not use the image data in the midst of a step. Should that test fix also be against `released` or should it be against `master`?

Also, if you've already started any of that or would prefer to be the one to work on it, just unassign me. Thanks!

omichel, 2023-02-08:
It should be against `master`. Any change in the code of Webots should target `master` (or `develop` if it's a new feature or breaking the ABI). Only the documentation changes with respect to the current released version (currently R2023a) should target the `released` branch.

omichel, 2023-02-08:
> Also, if you've already started any of that or would prefer to be the one to work on it, just unassign me. Thanks!

I believe @stefaniapedrazzi will be happy if you go ahead with a PR on this one.

brettle, 2023-02-10:
See issue #5870 for discussion of how/whether to address the underlying issue.

closed 2023-02-27 by stefaniapedrazzi