cyberbotics/webots #5803: robot_window_html test hangs intermittently

brettle, 2023-01-31:
**Describe the Bug**
Every once in a while the `robot_window_html` test hangs on my local machine.

**Steps to Reproduce**
1. Run `test_suite.py`
2. The suite will pause when it gets to `robot_window_html` with the robot window displayed in a new Chrome tab.
3. The test will eventually timeout before continuing.

**Expected behavior**
The test should pass consistently.

**System**
 - Fedora 37 Workstation
 - Graphics Card: NV167 / Mesa Intel® UHD Graphics 630 (CFL GT2)
 - Browser: Google Chrome 109.0.5414.74

**Additional context**
I suspect a race condition somewhere between webots, the robot controller, and the javascript running in the browser, but I've tried introducing intentional delays into the code in various intentional locations but I've thus far been unable to reproduce the problem reliably. 

Does this issue occur for anyone else?

omichel, 2023-01-31:
> Does this issue occur for anyone else?

Not for me, but it looks like a race condition indeed.

brettle, 2023-01-31:
Found it. Problem was occurring when I ran the tests while an instance of webots was already running. The test webots was listening on the same port and sometimes the original webots would be the one to accept connection.

This will be fixed by PR #5806.

closed 2023-02-06 by brettle