cyberbotics/webots #5809: Missing point on lidar

ShuffleWire, 2023-01-31:
**Describe the Bug**
When using high quality Lidar, I run into the issue that some points were not shown. i don't know if it a rendering issue of the Lidar by Wren, or a point cloud display issue, didn't tried to inspect that, but I'm confident about the display of the point cloud correctness.

**Steps to Reproduce**
1. Open the world and run one step
2. See missing point on the border

**Expected behavior**
All point should be displayed.

**Screenshots**
[image]

**System**
 - Operating System: Debian 11
 - Graphics Card: None, that could be a source of issue, please someone with one test for me :)  
 - Webots master 045000f6f723f93a57e34434dac05ff74a745013

ShuffleWire, 2023-07-26:
Still present in R2023b

omichel, 2023-11-06:
Fixed in #6417.

closed 2023-11-06 by omichel