cyberbotics/webots #5819: Unwanted warnings in distributed simulations

stefaniapedrazzi, 2023-02-01:
Following the recent changes to the contact points detection these simulations are printing many warnings that should be solved (tested on develop branch):
* `animated_skin.wbt`
* _continuing_

ad-daniel, 2023-02-01:
Perhaps we can improve the test worlds script so that it runs the simulation for some time instead of just checking whether there are warnings on load. A possible approach could be to graft a custom supervisor into the world file prior to loading it that runs the simulation for X seconds and quits

stefaniapedrazzi, 2023-02-01:
Yes, this would be good the detect the warnings.
However, we should always run the tests manually to check if the simulation behavior is still correct.

PR #6081 (2023-04-18): Fix contact points warnings
Fix #5819 and display more useful information in the contact point warnings allowing the user to understand which contacts properties should be updated to handle more contact points.
closed 2023-04-18 by omichel