cyberbotics/webots #5827: Where are the proto files?

thephet, 2023-02-02:
Hi There,

I am trying to get the proto files for the Scara T6 robot. The documentation says there are in:

    File location: "WEBOTS_HOME/projects/robots/epson/scara_t6/protos/ScaraT6.proto"

(https://www.cyberbotics.com/doc/guide/scara-epson-t6?version=develop&tab-os=windows)

But I have no protos file there.

I can see in the Github project you have them: https://github.com/cyberbotics/webots/tree/master/projects/robots/epson/scara_t6/protos/

Are these files anywhere in my computer?

I installed Webots using Linux (ArchLinux)

ad-daniel, 2023-02-02:
They are downloaded on a need basis when loading the world and subsequently cached into `.cache/Cyberbotics/Webots` but isn't not very human friendly. Instead, I suggest you to add the robot to your world, then right click the name in scene tree and select `Edit PROTO source` and save, which will save a local copy to your `protos` folder

thephet, 2023-02-02:
Hi @ad-daniel, thanks! that worked. I see I can also export to URDF format.

I have another couple of quick questions, as a Webots new user:

- Given the robot models in Webots, what would be the easiest way to get their measurements? For example, the diameter of the shaft.
- How can I integrate 3D models I design myself (like in Solidworks) with Webots?
- How can I export a Webots robot into a 3D cad software (like solidworks) so that I can design things around it? For example, if I want to design a gripper.

Thank you very much for your help

ad-daniel, 2023-02-03:
> Given the robot models in Webots, what would be the easiest way to get their measurements? For example, the diameter of the shaft.

That's a bit tricky, it depends how the robot was defined. Generally speaking we try to have external meshes for all components in .dae, .obj or .stl format. When that is the case, you can import these parts into solidworks and measure what you need to measure. For simpler geometries, usually webots primitives are used directly to define it, which means you need to dig into the PROTO and see the value set for these geometries. An easier approach could be to right-click the node and convert it to base-node in order to have access to this information directly in the scene tree. If the geometries are defined as `indexedfacesets` (which is usually the case for older models), then you'll need an extra step of converting this to something Solidworks will understand.

> How can I integrate 3D models I design myself (like in Solidworks) with Webots?

There is no one-click option, you need to create the different pars of the model in Solidworks, export them into .dae, .obj or .stl format and then assemble these components in Webots by adding joints & devices between the Solids. So when modeling it, you need to have separate parts for all components that could move independently. Anything else that has no moving parts can be exported as a single unit. Of course you'll need to configure the Physics of each solid to set the center of mass and mass of it, but this information you can usually get directly from Solidworks.

> How can I export a Webots robot into a 3D cad software (like solidworks) so that I can design things around it? For example, if I want to design a gripper.

As mentioned above, the full robot as a single unit might be tricky, but if you have access to the mesh of the component you're interested in, you can import that directly

thephet, 2023-02-03:
@ad-daniel Thank you very mucho for your details answers.

Is there any tutorial or similar to upload stl files into Webots?

closed 2023-02-03 by thephet

ad-daniel, 2023-02-03:
Not really, but it's pretty straightforward:
- add a Solid node
- in the children field of the Solid add a Shape node
- in the geometry field of the Shape add a Mesh node
- in the URL field of the Mesh node, point to the STL

Other than that, a number of basic tutorials are available here: https://www.cyberbotics.com/doc/guide/tutorials

thephet, 2023-02-03:
@ad-daniel thanks again!